marqs85/ossc #2: Need help for full compilation.

sorgelig, 2016-09-28:
Hi Markus,

I have experience with Quartus and have no problem to compile HDL projects. OSSC uses NIOS II and thus includes source code for NIOS firmware. I've tried to compile ossc.qpf project which doesn't involve NIOS firmware compilation.
Can you provide instruction how to compile OSSC firmware in Quartus? It seems some 3rd party IDE involved (Codelite?) which doesn't give the answer anyway (CodeLite doesn't provide the compiler).

borti4938, 2016-09-28:
Compilers are installed with the Quarts Prime software under the _installation_path/nios2eds/bin/gnu/H-x86_64-mingw32/bin_. However, you may use - as like as I do - Eclipse with NIOSII plugin, which is also installed with quarts prime. I hope that you are using Windows, where the setup is easier. I also use Linux Ubuntu Mate 16.04 where the setup of Quartus is quite awful.
You can start Eclipse over the Windows start menu or over Quartus under _Tools_->_NIOS II Software Build Tools for Eclipse_
Now you have to import two projects - the software and the board supported package (bsp). Just do a right click in the _Project Explorer_ and select _Import..._ -> _Import Nios II Software Build Tools Projects_. Select the root directory of the software and the bsp. Finally, you should have two projects.
To generate the BSP (has to be done the first time as well as every time you perform a synthesis of the quartus project) just right click on the bsp-project: _Nios II_ -> _Generate BSP_.
To compile the software just build it. You can run it directly on the hardware if you have a USB Blaster. To make the software ready for the firmware, make a right click on the project: _Make Targets_ -> _Build..._ -> _mem_init_generate_ -> _Build_.
Now you can rerun _Compile Design_ in Quartus Prime. Now just wait a bit until the process is finished.
To get the JIC-file go to _File_ -> _Convert Programming Files..._ -> _Open Conversion Setup Data..._ and select _ossc.cof_. The click on _Generate_. The JIC-file is then in the _output_files_ folder.
To build the firmware file for the SD card use the tool in the _tools_ folder of the OSSC project. The arguments for that are <input rbf file> <version> <version suffix>, so you'll use it e.g. like
./fw-tool ../output_files/ossc.rbf 0.73 myfw
I hope that helps :)

marqs85, 2016-09-28:
I've yet to write proper documentation, but borti4938's answer above covers basics when using Eclipse. Both BSP (software/sys_controller_bsp) and actual sw (software/sys_controller) contain Makefiles that you can use if you don't plan on using Eclipse. You need to install Nios2 EDS alongside Quartus II in any case to get compiler, linker etc. tools. Below are a couple useful commands you can run from shell:

Build BSP (need to do each time after RTL is compiled) (in software/sys_controller_bsp):
`nios2-bsp-generate-files --bsp-dir . --settings settings.bsp`

Build release version (in software/sys_controller):
`make`

Build debug version (in software/sys_controller):
`make APP_CFLAGS_DEFINED_SYMBOLS="-DDEBUG"`

Upload sw image and reset CPU. Open terminal if using debug version (in software/sys_controller):
`nios2-download -g --accept-bad-sysid sys_controller.elf && nios2-terminal`

Generate memory initialization file from sw image (in software/sys_controller):
`make mem_init_generate`
To update bitstream/programming files without full RTL recompilation and P&R, run "Processing->Update Memory Initializatio File" in Quartus II and re-run Assembler step.

sorgelig, 2016-09-29:
Thanks for info! I'm using Quartus in Windows, so it should be easy to setup (with info from above, of course).
It seems i've made mistake by ordering OSSC without Remote Control. None of my universal controls are working. I've found one code in universal control, which seems generates some recognized codes, but Menu code doesn't work :(
Marcus, i have logitech Harmony One remote, but it uses older app (new app isn't compatible), and there is no NEC E device to choose. Can you tell which exact TV mfg/model i need to choose to make it compatible with OSSC?

marqs85, 2016-09-29:
You need to first find a manufacturer code (try e.g. some Toshiba TV code) on your programmable remote that uses same protocol (not necessarily matching button codes) than OSSC - you know that if OSSC's green led reacts to remote control keypresses. After that, you can start OSSC in remote configuration mode by holding PCB button BTN1. You can now setup OSSC to match keyfunctions to any keys on your remote.

sorgelig, 2016-09-29:
I didn't notice OSSC can learn the RC codes. Now I have working RC.

rosaage, 2017-02-07:
Hello, I have made a DIY unit but I haven't gotten the audio to work as I can't get the firmware to compile.
When I try to compile with make APP_CFLAGS_DEFINED_SYMBOLS="-DDEBUG -DDIY-AUDIO" It stops on av_controller with:
`Info: Compiling ossc/av_controller.c to obj/default/ossc/av_controller.o
nios2-elf-gcc -xc -MP -MMD -c -I../sys_controller_bsp//HAL/inc -I../sys_controller_bsp/ -I../sys_controller_bsp//drivers/inc -Iit6613 -Itvp7002 -Iths7353 -Ispi_charlcd -Imemory -IulibSD -Iossc  -pipe -D__hal__ -DALT_NO_C_PLUS_PLUS -DALT_NO_CLEAN_EXIT -D'exit(a)=_exit(a)' -DALT_NO_EXIT -DALT_USE_DIRECT_DRIVERS -DALT_NO_INSTRUCTION_EMULATION -DALT_USE_SMALL_DRIVERS -DSMALL_C_LIB -DALT_SINGLE_THREADED -DALTERA_AVALON_JTAG_UART_IGNORE_FIFO_FULL_ERROR  -DDEBUG -DDIY_AUDIO  -Os  -Wall -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-function -fdata-sections -ffunction-sections -fshort-enums -fgnu89-inline  -mno-hw-div -mno-hw-mul -mno-hw-mulx -mgpopt=global  -o obj/default/ossc/av_controller.o ossc/av_controller.c
ossc/av_controller.c: In function 'SetupAudio':
ossc/av_controller.c:97:40: error: 'MODE_L2ENABLE' undeclared (first use in this function)
         if (video_modes[cm.id].flags & MODE_L2ENABLE)
                                        ^
ossc/av_controller.c:97:40: note: each undeclared identifier is reported only once for each function it appears in
ossc/av_controller.c:99:46: error: 'MODE_L3_MODE0' undeclared (first use in this function)
         else if (video_modes[cm.id].flags & (MODE_L3_MODE0|MODE_L3_MODE1|MODE_L3_MODE2|MODE_L3_MODE3))
                                              ^
ossc/av_controller.c:99:60: error: 'MODE_L3_MODE1' undeclared (first use in this function)
         else if (video_modes[cm.id].flags & (MODE_L3_MODE0|MODE_L3_MODE1|MODE_L3_MODE2|MODE_L3_MODE3))
                                                            ^
ossc/av_controller.c:99:74: error: 'MODE_L3_MODE2' undeclared (first use in this function)
         else if (video_modes[cm.id].flags & (MODE_L3_MODE0|MODE_L3_MODE1|MODE_L3_MODE2|MODE_L3_MODE3))
                                                                          ^
ossc/av_controller.c:99:88: error: 'MODE_L3_MODE3' undeclared (first use in this function)
         else if (video_modes[cm.id].flags & (MODE_L3_MODE0|MODE_L3_MODE1|MODE_L3_MODE2|MODE_L3_MODE3))
                                                                                        ^
ossc/av_controller.c:109:16: warning: implicit declaration of function 'read_it2' [-Wimplicit-function-declaration]
         cts |= read_it2(0x35) >> 4;
                ^
make: *** [obj/default/ossc/av_controller.o] Error 1
` It compiles fine without audio enabled.

borti4938, 2017-02-07:
First of all, why do you need the debug-options? You can leave them away.

Your problem is that [these four lines](https://github.com/marqs85/ossc/blob/release/software/sys_controller/ossc/av_controller.c#L97) are not aligned with the changes made with the latest commits (in particular [this one here](https://github.com/marqs85/ossc/commit/434186d64ef032a7d3b5965e4405a0b954cbe339).
You can work that around by replacing [line 97 to 100](https://github.com/marqs85/ossc/blob/release/software/sys_controller/ossc/av_controller.c#L97) with something like
`alt_u32 pclk_out = cm.fpga_vmultmode*(TVP_EXTCLK_HZ/cm.clkcnt)*video_modes[cm.id].h_total;`

rosaage, 2017-02-07:
I tried it and everything seems to work fine. It displays 1.76a on the screen now, but sadly no audio. Is there something I have to change in the menu, or should it automatically work?

borti4938, 2017-02-07:
You have to change the TX Mode from DVI to HDMI. Otherwise the IT6613E automatically mutes the audio. If this also does not work, try an older audio build as I'm not sure that my formula is 100% correct (my TVs are not that picky):
- official source: https://www.niksula.hut.fi/~mhiienka/ossc/fw/
- mirror: https://github.com/borti4938/Audio4OSSC/tree/master/fw-builds
(ooopps: I've seen that I missed to upload 0.75)
You can also send me some pictures of your installation of the mod by email.

marqs85, 2017-02-07:
Audio build is fixed in the latest commit.

closed 2017-04-06 by marqs85