marqs85/ossc #4: menu_sjis.c is in Makefile but missing in repository

alfishe, 2017-02-25:
ossc/menu_sjis.c file is not pushed and missing causing build error

marqs85, 2017-02-25:
menu_sjis.c is a Shift-JIS formatted menu.c that should be generated dynamically when building jp-lang image. If you're on *nix system, iconv does the trick:

make clean && iconv -f UTF-8 -t SHIFT-JIS ossc/menu.c > ossc/menu_sjis.c && make OSDLANG=JP mem_init_generate

alfishe, 2017-02-25:
Does it makes sense then to add it to correspondent target to Makefile as a dependency for mem_init_generate for JP enabled? Otherwise it's super-weird having a file registered in project/makefile but missing and without any idea why.

marqs85, 2017-02-28:
The Makefile was originally generated by Altera tools and contains lots of unnecessary stuff so it should be written from scratch with only relevant targets (which are currently defined in codelite project [file](https://github.com/marqs85/ossc/blob/release/software/ossc_sw.project#L163)

marqs85, 2017-10-28:
Fixed in 4ae3423acbb84ba359c6b3d83380b22061443225.

closed 2017-10-28 by marqs85