marqs85/ossc #12: PCB v1.6 audio

dwhinham, 2017-06-10:
Hi marqs,

I just successfully built a v1.6 PCB, which is very exciting - never assembled a project this complex before and everything appears to be working! Absolutely amazed by the beautifully sharp picture. 😄 

[image]

That is - except for the audio. I flashed the -aud version of the firmware but I don't hear anything over HDMI, even after moving the switch. I get audio output OK on the J6 jack.

Then I noticed that the PCM chip used here (PCM1862) is different to the one used in the audio expansion board (PCM1808) and appears to be software-controlled.

Does that mean there needs to be firmware support added before this chip will work?

Many thanks and best regards!
-Dale

marqs85, 2017-06-11:
Setup code for PCM1862 has not yet made its way into official fw since I received first prototypes not too long ago. I just pushed most recent changes here which include PCM1862 support (using same audio fw), so if you can build&flash the image yourself then you should be good to go. Otherwise, please send a PM/e-mail if you need a pre-built development image before the code makes it way into official image.

closed 2017-06-11 by marqs85

dwhinham, 2017-06-11:
I'm so sorry, I should have known better and checked the commit log, then I would have seen the PCM1862 code! Thankyou for the quick response.

I've got Quartus set up and will build and test tomorrow. 👍 
Cheers!