marqs85/ossc #22: Link profile with input and the other way around

borti4938, 2018-03-08:
From my intuition the def_input should be connected to the profile, too.
I do understand the 'link profile to input' in the following way:
- If I change the input, the last connected profile to this profiles loads. Good -> that works.
- If I now power cycle the OSSC I expect the input which was lastly selected -> no it's the def_input, which could be totally different.
- If I load a profile, I expect the last connected input to it to be loaded -> but this cannot happen in the current implementation.

So I would suggest to change the code as follows:
- option default input only appliable in link profile off
- store the active input to each profile as for the user hidden configuration
- this is somehow redundant to saving the profile number to input but allows for the two-way connection

borti4938, 2018-03-09:
I've written a draft to show what I mean:
https://github.com/borti4938/ossc/commit/e06b08f4ddfb46aac9730a5215cc85c035941753

My idea is to assume that the user uses a profile for a single input. This means you have optimized your settings according to the specific input.
So personally I expect
a) the profile to be loaded on input change, and
b) by loading a profile loding also the input I used for setting up this profile, too.

The draft is not fully tested. I quickly throw it into my OSSC, and tested loading profile, loading inputs and power cycle (last input and profile used). Everything works as I expect but so far I don't know if I changed every appropriate code portion properly.

What do you think about that approach, marqs? Maybe putting that idea up to a forum for broader discussions?

marqs85, 2018-03-10:
The original profile link code was done by megari (see #22), and we had some discussion on different approaches before ending up with the current implementation which we thought would be most straightforward for most users. Since there are various usage scenarios, I suggest adding a new profile-specific option (e.g. "auto-change input") instead of changing functionality of current options. With that, the following 3 options would control profile/input selection logic:

* Initial input (global)
   Selects power-on input, "Last Used" value fulfills your 2nd request
* Link input/prof (global)
   Links profile to input, fulfills your 1st request
* Auto-change input" (profile-specific)
   values: "none, AV1: RGBS, ... , AV3: YPbPr"
   changes input when profile is loaded, fulfills your 3rd request

Let's see what others think about this.

borti4938, 2018-03-10:
Yes, let's see where the discussion goes to and which ideas will come up. Your suggestion would be great for me either.

Maybe I was disappointed because I understood the profile / input link in the different direction than implemented. As written I'm used to switch between inputs by loading a profile - I had this implemented for me since a long time and made the decision with the 'default input' option which was profile specific. 

By the way, another good idea from Fudoh: copy profile function. Here also a draft: https://github.com/borti4938/ossc/commit/fe7450ee0d1028d21b4e34a3ae5eeb717cea0d11

borti4938, 2018-03-12:
I haven't realized that the v0.80 provides "Last Used" as default input. My fault... never switched to the very very end. However, it's interesting to see that all votes on shmups forum have ticked: boot into last av input. Although there were just five votes but I also hasn't read something different.
So I would put "Last used" as default option here.

For the rest, all voices seems to prefere somehow a different profile / AV input linking; no link at all, one way (either way) or two way. For me your suggestion seems the very best suite for all! So if you won't mind, I give it a try?

megari, 2018-03-12:
I think the suggestion @marqs85 made above is good and well thought-out. The initial suggestion for two-way linking, as it is worded, seems to contain quite a big flaw: it becomes impossible to change which one of the 10 profiles is linked to a certain input, as the input automatically changes if the profile index is changed (and vice-versa, although in that direction it is operation as originally intended). Therefore an option to disable such auto-switching is needed, and I think the default should be "auto-switching disabled". Of course, it depends on everyone's use cases whether they want it enabled or disabled, but disabled feels like a safer default.

A profile copy feature should be quite useful in combination with the profile+input linking options, although I am a bit worried about space getting tight on the device. Nevertheless, probably worth it.

megari, 2018-03-12:
Reading the shmups thread presented another use case: instead of primarily linking from inputs to profiles, we may also want to provide a primary linking mode of profiles to inputs. For instance, multiple profiles which, when selected, auto-switch to the same input. Maybe we could have this set of options:

- Initial input (global)
As before
- Input/profile link (global, many-to-one)
Values: [ None, Profile->Input, Input->Profile ]
- Auto-change input (profile specific)
Effectively a "Reverse input/profile link" when in input->profile mode, but one-to-one and secondary in nature to the above
Values: [ none, last used, AV1: RGBS, ... , AV3: YPbPr ]

borti4938, 2018-03-12:
Profile->Input link can be achieved with the third point; Personally I don't see the 'last used' option at the auto-change input option.

Throw away my initial suggestion. The copy function - as it was designed - could be used to work around of assigning a new input to the current active profile. But no matter... Discard it of your mind ;) 

Let's go with marqs suggestion

borti4938, 2018-03-12:
Could you take a look if I've overseen something?
https://github.com/borti4938/ossc/tree/profile_handling
(ca. 1.3kB margin for heap & stack is left in the aud-fw build)

megari, 2018-03-12:
Ah, indeed. Let me go through the different combinations of the last two options in @marqs85 suggestion, for each profile:
- Input -> profile link Off, Auto-change input none: No linking at all
- Input -> profile link Off, Auto-change input X (!= none): Effectively a one-way Profile -> input link
- Input -> profile link On, Auto-change input none: A one-way Input -> profile link
- Input -> profile link On, Auto-change input X (!= none): A two-way Input <-> profile link

The last option may require some care. If Input -> profile link is on and a profile X is assigned to an input Y, we must make sure that the profile X has Auto-change input either set to none or Y.

borti4938, 2018-03-12:
For the last point is my strategy to update the input->profile link (as implemented in current release branch) every time no matter if this link is active or not. In such a way you are up to date here if a user decides to go for link profile->input. And if not, just let the link win which was chosen: i.e. the input->profile if the input has been changed and the profile->input if the profile has been changed.

megari, 2018-03-12:
I'm trying to figure out the implications with regard to Flash wear. Do we end up potentially writing to the Flash every time we change the input or the profile? Maybe we're already doing that? I'll have to go and reread the code, as I haven't worked on it for more than a month.

megari, 2018-03-12:
@borti4938 Do we really need to write the initconfig to the Flash **every** time the input changes, instead of just when the input has explicitly been changed by the user?

marqs85, 2018-03-12:
I don't think the last point is a problem as long as "auto-change input" triggers only when profile is loaded manually (not via "Input -> profile link"). Same goes for "Input -> profile link", so profile is auto-loaded only when input is selected manually (i.e. not due to "initial input" or "Auto-change input").

megari, 2018-03-12:
I tried to implement the feature in a minimalistic way and created PR #23 for it.

borti4938, 2018-03-12:
OK, thank you...
What burned my mind was why loading the global profile number on init is enough even if profile link is active. But now I got it. :) From this (for me new) point of view, my implementation became quite redundant!
Thank you very much for implementing it. I'll take a closer look; but initially it looks very good :D

megari, 2018-03-15:
BTW, @borti4938 , sorry for hastily creating my own version and seemingly scrapping most of your work. There was no ill intent behind that. I just got excited that I came up with something a bit simpler and thought it would be easier to discuss it through a pull request. You definitely deserve the credit for the idea of extending the linking to a two-way one.

borti4938, 2018-03-15:
I'm happy that you jumped in :) It's very good discussing parts together.

borti4938, 2018-03-24:
Developed and discussed in #23

closed 2018-03-24 by borti4938