marqs85/ossc #26: Audio broken in 0.82

Shiftone, 2018-11-14:
I'm using a 1.6 board - coming from 0.78 to 0.82

After flashing ossc_0.82-aud.bin audio over HDMI no longer works. My sony Bravia reports "unsupported audio format"

PAL Sega Saturn connect over RGB Scart

Reverting to 0.78 fixes audio.

marqs85, 2018-11-15:
Any more details on the TV model? Also, could you try 0.79-0.81 to help pinpointing the change that might affect audio compatibility.

Shiftone, 2018-11-15:
https://www.niksula.hut.fi/~mhiienka/ossc/fw/ossc_0.79-aud.bin
https://www.niksula.hut.fi/~mhiienka/ossc/fw/ossc_0.80-aud.bin
https://www.niksula.hut.fi/~mhiienka/ossc/fw/ossc_0.81-aud.bin

0.79 - 0.81 audio working

https://www.niksula.hut.fi/~mhiienka/ossc/fw/ossc_0.82-aud.bin

0.82 broken

Sony Bravia kdl-40w605b

https://www.sony.co.uk/electronics/support/televisions-projectors-lcd-tvs/kdl-40w605b/specifications

Shiftone, 2018-12-01:
Any progress on this bug?

marqs85, 2018-12-03:
I checked that there was no audio related changes between 0.81 and 0.82 so I guess this might be related to timing. Have you tried enabling "Full TX setup" in compatibility options?

Shiftone, 2018-12-05:
Full TX is enabled

Another quick round of tests yielding the same results.

Tested with a couple of other devices as well just to make sure.

Sega Saturn RGB Scart on AV1
Sega Dreamcast RGB Scart on AV1
Sega Dreamcast VGA in AV3 with RCA audio to A3 In

same results

No sound on 0.82 - works on all previous firmwares

marqs85, 2018-12-05:
I2C transactions are actually hw-controlled, but any delays between transactions and for other controls (e.g. HDMI TX chip) are generated using SW delays which indeed might have changed a bit due to CPU update. @Shiftone have you tried v0.82 on another TV? That could help narrowing down whether the cause is in PCM1862 control or HDMI TX control.

marqs85, 2019-01-10:
I found a bug in the code where uninitialized data ends up in the audio infoframe. I made a test [firmware](http://www.niksula.hut.fi/~mhiienka/tmp/ossc_0.82-aud-t.bin) with the fix, feel free to test in your setup.

Shiftone, 2019-01-22:
Hi marqs, just got round to testing the firmware above. Same result, unsupported audio format. Rolling back to 0.81 fixes.

I'll try and test on another display this week if I get an opportunity to grab a console and take it downstairs.

Shiftone, 2019-03-17:
Ok, again, sorry for the epic delay.  Just got round to testing my Mega Drive on the OSSC on another screen. Audio works fine in 0.82 on my Toshiba.

So, whatever broke, broke in a way that this Sony Bravia very specifically doesn't like.

marqs85, 2019-09-01:
0.84 has a few additional fixes on audio Infoframe which might help here.

JRLS, 2021-02-19:
same problem here.
I went to 0.82, and audio is broken on LG oled C7 tv, which is not a cheap tv.
had to revert back to 0.81.
I hope this can be fixed, otherwise this ossc is becoming useless to me...

marqs85, 2021-02-21:
What makes this challenging is that I can't reproduce the issue in my setup and the fact that there are no direct audio related changes between 0.81 and 0.82 as mentioned earlier. I can only assume the problem has appeared on some of the various commits between 530df66 and 4c1b211. I rebuilt some of intermediate versions if you want test and help pinpointing where it has stopped working:

[ossc_0.81-rebu.bin](https://www.niksula.hut.fi/~mhiienka/tmp/ossc_0.81-rebu.bin): v0.81a rebuilt - should work just like exisiting v0.81.
[ossc_0.82-4676cbd.bin](https://www.niksula.hut.fi/~mhiienka/tmp/ossc_0.82-4676cbd.bin): rebuilt from 4676cbd - fw cannot be upgraded via SD using this version - DO NOT FLASH this unless you have a USB Blaster avaiable.
[ossc_0.82-7914a2e.bin](https://www.niksula.hut.fi/~mhiienka/tmp/ossc_0.82-7914a2e.bin): rebuilt from 7914a2e - fw upgrade from SD is operational.

JRLS, 2021-02-21:
hi Marqs, 

1) but, you said:
"I2C transactions are actually hw-controlled, but any delays between transactions and for other controls (e.g. HDMI TX chip) are generated using SW delays which indeed might have changed a bit due to CPU update"
So, there might be a software problem?

2) Have you tried it on a LG oled C7 TV with firmware 0.82 yourself?

3) anyway, i can test ossc_0.82-7914a2e.bin pretty soon and see what happens

4) i havent tried ossc_0.88-aud.bin yet, but probably has the same problem?

JRLS, 2021-02-21:
also, you said to shiftone: "have you tried v0.82 on another TV? That could help narrowing down whether the cause is in PCM1862 control or HDMI TX control."

shiftone has tested it on toshiya tv, and sound works, so what is your conclusion on why the sound does not work on his other TV? Is the problem in PCM1862 control or HDMI TX control ?

marqs85, 2021-02-21:
1. SW problem in the sense that the same application code is behaving differently when compiled for another CPU core.
2. I don't have one to try, unfortunately. I have not heard reports from other LG OLED owners, though - majority of reported audio issues have been with Sony TVs.
3. Ok
4. Most likely it won't be any different to 0.87-aud.

It's hard to guess why this happens with a few particular TVs. The audio stream itself is probably just fine, but there might be some issues with metadata even though that has been checked a few times (not using a proper HDMI analyzer, though).

JRLS, 2021-02-21:
thanks a lot, 
i updated firmware with ossc_0.82-7914a2e.bin:, 
it says, firmware ok, verify ok, now reboot oscc.

i reboot

the oscc is now dead!
red led on, no display
thank man

JRLS, 2021-02-21:
the light of the lcd display is on, but nothing is showing

JRLS, 2021-02-21:
how can this be fixed?

marqs85, 2021-02-21:
Send an e-mail to the address shown on top of source files and let's see what can be done.

JRLS, 2021-02-21:
on top of source files?
what does that mean? i dont understand?

marqs85, 2021-02-21:
[This](mailto:mhiienka@niksula.hut.fi) address.

JRLS, 2021-02-21:
i bought my ossc with videogameperfection
can i return it for repair?

marqs85, 2021-02-21:
The easiest way is to sort you a USB Blaster programmer which you can use to restore the firmware. You can contact me directly in this case.

JRLS, 2021-02-21:
how can i contact you directly? via email?
so, you're saying that it can be fixed with an usb blaster programmer?
and how does that work?

JRLS, 2021-02-24:
ok, reflashed with terasic usb blaster and 0.81 jic, working again (*pfew*)
still, the audioproblem from 0.82 and up remains....