marqs85/ossc #28: 8.2 and 8.3: Can't replicate VGA settings from 8.1

stephen-usher, 2019-06-15:
In 8.1 I required the following settings for the VGA output from an Atari TT to correctly display the output:

Sampling Opt->640x480 Adv Timing:
H.Samplerate: 911
H.Backporch: 58
V.Backporch: 31
V.Sync: 1

Allow upsample2x On

Output Opt->480p/576p Proc. Line 2x.

I've tried many combinations of settings after upgrading to 8.2 and thence 8.3 but none will expand the display horizontally as the 8.1 version "H.Samplerate" allowed. No setting alters the display width to cope with the pre-standard timings.

marqs85, 2019-06-17:
Do you have "480p in sampler" set to "Auto" or "VESA 640x480@60"? Your comment implies that H.Samplerate has no effect at all which shouldn't be possible unless that option is wrong. There's no changes between 0.81 and 0.82 that affect how sampling works.

stephen-usher, 2019-07-14:
Sorry for taking so long to reply. I've not had a chance to check until now as I was preparing machines for an exhibition and didn't have space to set up the Atari TT.

It seems that I did have that setting set. It's rather confusing that the option to adjust the value appears in the menu if the sampler setting is "Auto" and is hence not functional.

marqs85, 2019-07-15:
I'm not sure if I understood correctly, but I presume the issue was "480p in sampler" having been set to DTV-480p which caused 640x480 Adv Timing adjustments end up with no effect? With VGA (AV3) input, "Auto" and "VESA 640x480@60" values activate 640x480 mode preset while "DTV-480p" activates 480p mode preset. You then have to make sampling adjustments to the correct preset, which is also pre-selected on Adv. Timing menu if you are running the system while before starting to make adjustments.