marqs85/ossc #29: Feature Request: 400p PC mode with Line2x and Line3x

James-F2, 2019-07-30:
Greetings Marqs.

CRT and 4:3 LCD monitors are becoming rare, and I think the OSSC can do great job for supporting retro PCs.

Currently the OSSC lacks a proper 400p mode support for retro (DOS) computers that output 720x400 text mode as native resolution via VGA.
The OSSC handles 400p resolutions in the 384p menu (confusing) and lacks a much needed Line3x.
Proper 4:3 aspect ratio can be achieved by adjusting the Samplerate and lowering Active pixels, but Line2x is simply not enough sampling rate for smooth pixels with native 720 horizontal resolution so the pixels are deformed, that's why Line3x is much needed in 720x400p modes.

Current 640x400 need to be renamed to 720x400 and default timing need to be changed as in the link below, it will be pixel accurate (native) like connecting directly to a monitor with VGA cable, but will not result in 4:3 image.
To produce a 4:3 image, a custom timing with Line2x or Line3x is needed with lowering the active pixels.
Further info I've posted on VGP forum:
https://www.videogameperfection.com/forums/topic/400p-mode-for-retro-pc/#post-27188

I hope too see more love for 400p mode and DOS gaming on the OSSC.

Thank you,
James

James-F2, 2019-07-31:
You were right.
640x400 and 720x400 have different clocks form my retro video cards.
I used a 320x200 (640x400) grid test pattern and it looked pixel accurate only with 800 samplerate, most if not all games use this.
Text mode 720x400 only looks pixel aligned with 900 samplerate.

Lin3x would greatly help with low sampling rate and deformed pixel in 4:3 image.
I'm wondering of something can be done with Line2x to help produce 4:3 image in the standard 1280x800 resolution?
Thank you again for looking into this.

marqs85, 2019-08-02:
400p support improved in 4a686da.

closed 2019-08-02 by marqs85