marqs85/ossc #30: Loss of remote-control functionality after updating to firmware 0.83a

DeviousMalcontent, 2019-08-12:
After completing a firmware update, my OSSC remote control no longer works. – the Green LED on the front of the OSSC unit flashes to indicate input, but nothing is happening, I can only change AV inputs by pushing the two BTN0 and BTN1 on the front of the unit. - Any advice? I am on firmware 0.83a.

marqs85, 2019-08-12:
Do you have the standard L336 remote with pre-programmed keys for OSSC? If not, you need to reconfigure button mappings after fw update by holding down BTN1 while powering on OSSC.

DeviousMalcontent, 2019-08-12:
I do believe that I have an L336 remote with pre-programmed keys as it is this exact model of remote control pictured: https://www.videogameperfection.com/products/converter-control-overlay/

That said, Holding down the BTN1 before powering on the device has allowed me to remap the buttons to this remote control and thus fixed the issue. 

Thank you very much! :D

closed 2020-06-08 by DeviousMalcontent