marqs85/ossc #34: Audio cuts in and out on certain settings

GarlicToum, 2019-10-13:
I'm a big fan of the OSSC, and recently got some Carby Component cables for my Game Cube. I've noticed the audio cuts in and out occasionally with various settings on the latest firmware. This same behavior also happened with a PS2 with component a long time back (definitely an older firmware version), and still happens with the PS2 via SCART. Here are the settings that cause me some issues:

 - 480i line 4x (audio cuts every 5-10 seconds).
 - 480i line 2x or 576p line 2x (minutely).

Whenever the audio cuts out, it comes back almost immediately and seems to be tied to the input processing. It doesn't cut out with passthrough.

My TV (Sony XBR55-X900E) doesn't really display 480i Line 3x (a lot of flickering), so I didn't test it. I have not yet tried a 960i/1080i input. The same thing happens when switching to different audio cables on the Game Cube, switching to a different Game Cube or PS2, or using a Wii (w/component out + analog audio). All the rest of my setup (SNES w/dejitter and N64) uses SCART cables and the audio always works great regardless of scaling settings.

I have tried various compatibility settings to no avail:

- TX Mode
- HDMI ITC
- Full TX setup 
- Panasonic hack (even though it's a sony)
- Downsampling on/off
- swap left/right
- ADC gain

Any help/ideas would be appreciated!

marqs85, 2019-10-14:
Are 240p line2x and 480p line2x then OK? Could you test the problematic modes with another display just to rule out some possible causes?

GarlicToum, 2019-10-15:
240p line 2x is fine. I couldn't specifically test 480p line 2x (I'll continue looking for a game) vs 576p line 2x which does cut out.

I plugged the OSSC into a monitor with sound and the audio worked fine without cutting out.

I also tweaked all the sound settings I had on my TV and disabled the soundbar, but that didn't seem to help.

-------- Original Message --------

GarlicToum, 2019-10-20:
So, I tried out some things on the GameCube using Swiss to force certain outputs and noticed 480p line 2x still causes the audio to cut in and out. If I set it to 960i line 2x (bobbed) there are no audio issues (this is using Carby component cables).

I'm going to try and get a hold of a PS2 game that supports 1080i and see if that exhibits the same behavior as 960i forced GameCube.

GarlicToum, 2019-10-23:
I did get a hold of a PS2 game with 1080i support. In line 2x (bobbed) there were no audio drop outs.

GarlicToum, 2019-10-26:
Just tried out the gamecube with 480p passthru and it works fine. I also updated the firmware on the tv and that didn't seem to do anything.

marqs85, 2019-10-28:
So the problem seems to be tied into certain modes. One thing worth trying with 480i x2 is to set "Default HDMI VIC" to 2 under compatibility opt. before selecting the source on OSSC. That should also make metadata identical to 480p passthrough just in case the TV's decoder relies on that.

GarlicToum, 2019-10-28:
I gave that a go and it also didn't seem to work...

GarlicToum, 2019-12-04:
Thanks for taking the time to look into it marqs. I'll figure something out to work around the audio issue (either an extractor or a receiver of some kind). 

Also, a big thanks for making and developing the OSSC, it is simply the best for retro video gaming.

Feel free to close this ticket, if something comes up in the future, I am willing to test it out.

PapiFed, 2020-03-31:
I would like to report I am experiencing the same audio cuts problem reported here.
More specifically when using 480p x2 with H.Samplerate value of 858.00

If I switch to 480p passthrough (still maintaining H.Samplerate value of 858.00) audio works perfectly fine.
If I stay in 480p x2 mode and change the H.Samplerate value audio works perfectly fine but then obviously I get a blurry signal.

I am using a Sony 65XE9005, could it be a Sony thing ?

marqs85, 2020-04-01:
Audio issues (not just this one) indeed seem to be much more common with Sony's devices than with others. I'll look into it if I have a chance to test some of these problematic models in future.

PapiFed, 2020-04-02:
Thanks for getting back to me, much appreciated.

lucasbrock84, 2020-05-27:
Hey. I’m new to the OSSC world, and similarly am having audio cutting issues with a GameCube, running Carby cables, to a Sony XBR49X900E. Trying to run at 480p doubled.

PapiFed, 2020-06-03:
Several other users reporting the same problem on [VGP forum](https://videogameperfection.com/forums/topic/sound-drop-sync-issue-tv-issue/).

tmenzzz, 2021-04-16:
sony x900h is also doing this. 480p 2x has audio dropouts.

marqs85, 2021-04-19:
Does this issue occur on firmware v0.81? Some people have reported audio incompatibilities on later fw versions which might be related to this.

lucasbrock84, 2021-04-19:
Marqs, thanks for responding. My OSSC says firmware 0.82a. 

Luke

danilodequeiroz, 2021-04-25:
I 'm having the exact same issue with my OSSC on a TCL 49c2us, when outputting 480pX2 or 480iX3, the audio cyclic cuts, the only workaround is to use respectively pass through for 480p and 480iX2 instead. The issue doesn't happen on my secondary LG 1080p TV which looks like to be some way "tolerant" to this audio signal anomaly.

GarlicToum, 2021-04-27:
@marqs85, I went ahead and tried 0.81, and had the same audio cutting in and out.

retrocats, 2021-04-27:
i know sony x900h was already mentioned, but just adding that i have the same issue with the same tv. using dreamcast, with vga cable.

JomSpoons, 2021-07-20:
Also have a Sony TV and this issue occurring. Would appreciate if a fix could be worked out in the foreseeable future, it gets distracting during gameplay.

siwy102, 2021-08-10:
Hello,
Also have sony TV (43XE7005) And the same issue with audio cut offs, 
using ps2 pal, x1 x2 works fine, x4 cuttong off audio
xbox classic ntsc x1 works fine, x2 cutting off
Gamecube x1, x2 ok, x4 not ok
As i noticed in x4 mode in game gran turismo 4 it dropping off audio in specific moments, on fast corners and near audience, yes really! I hope it will help thanks.

PapiFed, 2021-09-02:
Not sure if this can help in any way but using a Retrotink 5x does not present this problem.

marqs85, 2021-09-02:
@siwy102: so there is some relationship between audio frequency and occurrence of this issue? Sounds very strange, but I had similar problem with a PC sound card 20 years ago.
@PapiFed: it doesn't really help unless RT5x outputs exact same timings which I highly doubt. It'd be to good if someone could create a custom modeiine with PC with matched 960p timings and test if the issue replicates. That would confirm it's a problem with how Sony sets handle this mode.

PapiFed, 2021-09-02:
@marqs85 Is this not what we already tried last summer ?

https://videogameperfection.com/forums/topic/sound-drop-sync-issue-tv-issue/#post-38528
https://videogameperfection.com/forums/topic/sound-drop-sync-issue-tv-issue/page/2/#post-38571

TsukikoChan, 2023-12-14:
Hi, sorry to res this old bug but was a fix or mitigation ever found for 480x2 on sony tvs? 

To add to the old thread I have a 55x900e as well, ossc on .89a fw (1.5ossc with audio mod), packapunch rgb dreamcast cable using 480p setting, pal dreamcast and i get audio drops.

lucasbrock84, 2023-12-14:
Tsukiko-san, I had not found a fix before my 18 month old son took a statue to our 49” X900E. The TV is now sadly in the trash. Best of luck finding an answer. Luke

marqs85, 2023-12-14:
Has any Sony owner tried the [v1.xx series](https://videogameperfection.com/forums/topic/ossc-v1-xx-series-beta-firmware/) firmware/mod?

TsukikoChan, 2023-12-15:
I haven't yet, i need to do the soldering on mine first and then i can try the firmware. I'll be able to over the xmas period and i'll report back for my dreamcast 480p2x