marqs85/ossc #37: Square pixel aspect ratio

alessandropaggi, 2019-11-13:
Will it be possible to add the possibility to select the output aspect to maintain square pixel aspect ratio for resolutions other than 256x240?

marqs85, 2020-04-27:
Can you elaborate a bit which resolutions you mean, 384x240 and 512x240 perhaps?

alessandropaggi, 2020-04-27:
OSSC menu has "256x240 aspect" menu entry, that allows for 256x240 optimized mode to select between 4:3 and 8:7 aspect, the latter to maintain a square pixel ratio.
It would be nice to have the possibility to select an aspect the yields a square pixel ratio also for the other modes, that is 512x240, 384x240 and 320x240.

marqs85, 2020-04-27:
320x240 preset is already processed as square PAR. It'd be possible to add such option for 384x240 and 512x240 in certain output modes at least, but I'm not sure where that would be useful (video capture perhaps?).