marqs85/ossc #39: Ability to map Y/Pb/Pr inputs

Namaneo, 2019-11-29:
Hello,

After some research, and knowing that the OSSC can handle YPbPr through scart, how hard would it be to develop the ability to map each of the three inputs?

Let's take the example of the Wii : to force the console to output YPbPr, there's only two pins to jump on the motherboard (links below). However, in this mode, here's what happens:

* Y is sent through the red track
* Pb is sent through the green track
* Pr is sent through the blue track

YPbPr through scart seems to work that way instead:

* Pr is sent through the red track
* Y is sent through the green track
* Pb is sent through the blue track

Would it be possible to have settings allowing to map each YPbPr track to each RGB track, for this kind of purpose?

All said above is assuming that my sources are correct. Please tell me if I'm wrong or if I missed something !

Links:

* Wii motherboard: https://bit.ly/2Y93xzX
* Wii to Scart: https://bit.ly/37OrVeu
* Scart mapping: https://bit.ly/2svHtUj

marqs85, 2019-11-30:
Only Pb/Pr could be theoretically swapped as Y line carries sync and thus splits to video and sync inputs of the digitizer chip.

Namaneo, 2019-11-30:
So there's no way to configure something like "Y comes from this SCART pin, Pb from this one and Pr from this other one"?

marqs85, 2019-11-30:
Yes, it cannot be done without HW modification.

Namaneo, 2019-11-30:
Alright, too bad
Thanks for answering anyway !

closed 2019-11-30 by Namaneo