marqs85/ossc #40: Suggestion for lat_tester.v: report 10% and 90%  thresholds

alanbork, 2019-12-15:
I've been experimenting with the latency tester and I'm getting weird values and lots of false triggers. I read the lat_tester.v code and can't really make sense of it, in part due to the variable names. But stepping back a bit, here's a suggested algorithm that might be more robust.

1) measure the brightness of the white static probe box
2) measure the brightness of the black screen
3) display the probe
4) report the time for the brightness to rise to 10% of the distance between black and white.
5) report the time for the brightness to rise to 90% of the distance between white and black.

There's enough space on the lcd to display both 4 and 5 simultaneously. Also, I think the 10% and 90% capture the nature of latency nicely:  how quickly the display starts to reflect the changed signal (10% brightness), and how long it takes to reach the final state (90%). I suggest 10% and 90% instead of 5% and 95% for better noise rejection.  If there's enough space on the LCD you could fit in more percentages, but it seems like the top half is reserved for reporting the mode? 

Is this too complex for the system to run in real time? At its simplest it's just a matter of setting the threshold based on the pre-test probe.

marqs85, 2019-12-16:
The sensor output is connected to a digital input pin so it can be read only as a binary value. Threshold of the sensor can be tweaked by varying the resistor value which could help with false triggers (i.e. when sensor triggers on a black screen due to backlight leakage etc.), but otherwise there's limited number of things that can be done. If you can probe the sensor output on a scope, you should be able to see the frequency and length of the signals causing those false triggers. If they are short enough, then simple low-pass filtering (on FPGA) could be added to take care of them. I presume your issues have been with LCD since the tester has worked quite consistently with plasma & OLED?

alanbork, 2019-12-16:
Yes, it's an older Sony LCD. I used the 150k resistor suggested. Now I
understand that bit about using a potentiometer in the instructions for
building one. Could I just wire a pot in series with the photodiode?

What's the meaning of the two MS values printed? I assume the first one is
the lag but the 2nd is mysterious.

alanbork, 2020-01-06:
A easy alternative to a pot is just sliding a index card in front of the photo-diode. Thus I can limit the amount of light reaching the sensor, and prevent back-light fluctuation from causing a false trigger. But the amount of light I block changes the lag values reported significantly. Presumably I am measuring the different parts of the probe brightness onset curve.  It would be nice if there was an option to start the measurement when the photo-diode threshold trigger was reached, or in a loop that was triggered every 5 seconds or something like that. Otherwise it's very tricky to adjust the threshold (via my card method or a pot).

still wondering what that 2nd value reported on the lcd means. It shows up as about 2-3ms on my display.

alanbork, 2020-01-06:
Here are some hard numbers: by adjusting the amount of light hitting the sensor I can measure "lag" values of 25 to 35 ms in the top-left. In the middle its 20-35ms. And the bottom shows 24-35ms. Clearly I'm not measuring the painting of the frame here, but variations in how bright the back light is. All the more reason to have an auto-trigger mode. 

Perhaps a simpler option: an indication on the OSSC LCD if the photo diode response is above the trigger level so you could adjust your rig (by whatever means, pot or???) to consistently be just above that trigger level when the probe is visible.

marqs85, 2020-01-06:
Binary status of the diode could be shown on the screen to help setup, but it wouldn't reveal if there's noise/glitches on the signal which may occur close to the threshold.  A measurement loop and averaging could be easily added on the implementation. The second ms number is the length of the light strobe, but that number shouldn't pop up with sample-and-hold type of displays. Perhaps your LCD backlight uses low-frequency PWM which would explain that - have you tried measurement with its backlight set to maximum (PWM duty cycle = 100%)?

alanbork, 2020-01-09:
The pwm idea is quite plausible, however I just set back-light brightness to max and it didn't change the values observed. My 2nd value on the display varies from 1 to 4 ms. For instance, the slowest typical lag recorded is 34.4ms/3.3ms.  Of course, it could be strobing on each frame to reduce motion blur. That actually seems moderately plausible. 

The averaging idea is nice, though at the moment it is helpful to see the raw values to help diagnose what's going on. 

Here's a 2nd idea which might at the least offer a sanity check on the photodiode measurements.  The ossc has a LED; how about turning that LED on at the moment that the probe image is sent to the TV? Then a decent slow motion camera (such as any medium to high end smart phone; my 4 year old flagship does 800fps) could measure the lag between LED and probe appearance. This idea, at least, should require very minimal changes to the code. Bonus: it would allow lag measurements even if you don't attach a photodiode to the OSSC.

marqs85, 2020-01-21:
I had a chance to measure various low and mid end LCD monitors yesterday with a scope attached as well. That confirmed some of my suspicions mentioned earlier, but also brought up a couple surprises.

Firstly, backlight PWM frequency (typically 150-200Hz) on these monitors was low enough to toggle the sampled sensor output, and in many cases the frequency wasn't even synchronized to refresh rate. It means that unless you have a monitor with flicker-free backlight, you should max the brightness to get 100% PWM duty cycle which provides most consistent results.

Even with max. brightness, sensor output for some monitors ended up hovering in ~1V range during white pattern instead of getting fully pulled down to 0V. That still crossed the input low threshold which enabled measurement, but to get a reliable reading on these kind of dim monitors, max. brightness is a must unless the pull-up resistor value is tweaked.

For certain monitors the measurement failed due to the fact that they triggered the sensor on a black spot. Setting brightness from max to min didn't help although then sensor triggered at the rate of PWM instead of being fixed at (triggered) 0V output. It seemed like these monitors leaked light badly enough even though you couldn't tell that by naked eye. Perhaps the light was outside of visible spectrum - SFH-300 has spectral range of 450-1000nm so it includes near-infrared area as well. It might be worth trying to swap in another photodiode with narrower spectrum sensitivity.

alanbork, 2020-02-06:
That's interesting indeed. I wouldn't expect much emissivity outside the visible spectrum but you never know. Back when I had a job I used a research grade spectrometer for calibrating monitors for color vision experiments and the LED backlights were relatively tight in their spectra, very little IR or UV.

I still wonder if the photo diode approach is relatively doomed though as long as the output is measured with a binary threshold.  While it might be improved with some tweaks you describe there's a lot of potential sources of false triggers. As I mentioned in my last post even when setting backlighting to max (presumably, no PWM) I still get a lot of variability, and probe durations of only 1-4ms suggesting my set at least still pulses the backlight. 

What's the chance of implementing the fallback option I suggested of setting the LED on the OSSC to be turned on when the probe is sent to the TV so that external measurement methods (like a slow motion camera) could also be used either in lue of the photodiode all together or as a sanity check on the photodiode numbers?

marqs85, 2020-02-09:
An uploaded test firmware is available [here](https://www.niksula.hut.fi/~mhiienka/tmp/ossc_0.86-test.bin) which uses LEDs for debug in latency tester mode. Green LED indicates the status of the sensor while red LED is ignited when first white pixel is scanned out and turned off when first trigger occurs.

I also bought a bunch of SFH 3310 phototransistors to test if they make any difference with the problematic displays.

Sirethief2525, 2020-02-10:
Hey marqs85, I know this doesn't pertain to the actual conversation. But I
was wondering if somehow a firmware update would be able to make the ossc
output a standardized 720p resolution. I know that it's a line
multiplication device, but perhaps that would help with some compatibility
issues with some TVs and capture cards. Thanks in advance, Steve.

alanbork, 2020-02-14:
fantastic!

I just got a chance to play with this and it works great. Here's my results. My camera (samsung s7) is slower than I recalled, just(!) 240fps, for a resolution of roughly 4ms.

holding the camera in portrait mode I consistently find that after the LED lights up 5 frames elapse before the center probe rectangle is clearly visible, at least 50% brightness.  that's 20ms.  I ignore the occasional frames where the slightest outline of it is visible on the 4th frame.

in landscape mode the results are 1 frame slower, working out to 25ms lag. I'm guessing that's a rolling shutter issue (the OSSC is sitting below my TV, so depending on which way the CCD/whatever is read the results could be off by one frame count).

moving to the lower corner, also landscape I consistently measure 6 or 7 frames of lag (25-29ms) before full brightness of the probe. Interestingly, the sampling is such that I start to see the probe outline 1 frame early pretty much every time, maybe 10-25% of full brightness.

All the while you can see some evidence of the backlight strobing; the overall brightness of the screen goes from high to low, alternating each frame. So I guess that's roughly a 120hz strobe? that seems slow....

so to summarize: the slow motion camera option gives fairly easy to read results with a good bit of sanity check in them (makes it easy to avoid counting the leading edge of the probe impulse, and backlight flicker). using the photodiode I got typical lag values of 25 to 34ms, once I elmiated the worst of the false triggers by physically reducing the sensitivity of the diode and setting the backlight to max. but sometimes I also got 50 or 80 ms lag (presumably because I reduced the sensitivity too low. The camera approach, because you can actually see what's going on, is more robust AND has the advantage that no SMT soldering skills are required ;-).

Note (to self, mostly) the camera approach could be even better than what I did here - using a tripod to record the data one could then import the data into matlab/octave and plot the brightness of regions corresponding to the LED and the probe/backlight and measure the phase shift, allowing temporal resolution better than the absolute sample rate of the camera. Awesome!

alanbork, 2020-02-14:
quick clarification: it sounds like you are turning the LED on when the probe pixels are actually sent to the monitor, rather than at the start of the frame. Correct? I assume HDMI scanout is like VGA; the pixels are sent just in time? it would be easier to explain the difference I saw in lag between middle and bottom probes if the LED is actually lit at frame onset rather than probe onset, but of course there's also whatever processing the TV does to confuse things.

marqs85, 2020-02-14:
Yes, measurement starts (and LED turns on) when first white pixel is sent to HDMI transmitter, not at the start of the frame. If the results are different between the 3 points, it just tells the monitor does some buffering and scans out at faster rate (e.g. 120Hz).

alanbork, 2020-02-15:
Thanks for that clarification. If you decide to merge this into the main release may I suggest one more easy change: add a 4th probe style, a single bar the same width as the current probe, but running from the top of the screen to the bottom? Since it's possible to capture the entire screen with a camera this would provide an easy way to estimate how far down the screen the probe has been "painted" on each captured frame, and whether the occasional dimness of the probe is due to scanout, backlight pwm, pixel response time, etc.

alanbork, 2020-02-16:
I tried this with an ancient panasonic plasma TV (505PU). The photodiode showed 28ms/top, 20ms mid, 14ms bottom (suggesting faster than 60hz scanning). I guess that the photodiode method also measures lag from the first white pixel "sent". I understand why you did it this way, but it is confusing compared to all lag testers out there. Hey, your code, your decision.  I'll update the Wiki so that more people understand what the lag OSSC numbers are telling you, though.

Meanwhile, the camera method worked great again. This particular display has a pretty slow black-white transition; for instance on the top position, the first hint of the probe was seen after 20ms, but it didn't reach full brightness until roughly 30ms, very much in line with the OSSC photodiode measurement. Of course, a more sensitive diode or brighter backlight might have triggered at 20ms.  The green light mod is also super helpful in determining how  sensitive the diode is, what the cause of false triggers is, etc. Very nice work!

marqs85, 2020-02-16:
The counter starts as well from the first while pixel sent. I see this as the correct way to measure _display latency_, i.e. the time display adds on the whole latency chain. The finite scanning speed matters in big picture only if the source frame comes from a framebuffer which is not updated during scanout (vsynced) - I'd never include that number in display latency value.

About plasmas, their sub-field drive operates at much higher frequency (e.g. 600Hz) which is why the lowest position has least latency

alanbork, 2020-02-24:
It is your choice as the designer, for sure.  I don't think everybody measures it that way, however, so it's worth being clear in the documentation, which appears to be the wiki. 

Thus, I've updated the wiki on junkerhq to be more clear about usage on the currently official firmware.  Is this testing firmware you've uploaded going to be part of an official release? If so, any chance you could implement the vertical bar testing stimuli I requested?

marqs85, 2020-02-28:
The changes in testing firmware will be included in the next official release. I'll consider the vertical bar suggestion as well.

alanbork, 2020-05-18:
Just to close the loop, I went ahead and implemented this idea  (the long
vertical bar) using a raspberry pi. works great.
https://alantechreview.blogspot.com/2020/05/a-5-tv-input-lag-tester-using-raspberry.html

alanbork, 2021-07-03:
the fact that actual 10%/90% thresholds could not be measured eventually lead to me making my own tool with a proper ADC that attaches to a rpi:

https://alantechreview.blogspot.com/2020/08/pilagtester-pro-order-page.html

closed 2021-07-03 by alanbork