marqs85/ossc #41: Feature request: Audio panning

rvalles, 2020-03-20:
Some sources (namely the Amiga) have dedicated audio channels for Left and Right.

It would be nice to support _optionally_ panning the audio so that all channels are present on both sides.

This could be accomplished by making each output channel a sum of Left and Right, with great care to prevent clipping, ideally with an adjustment of how much to cross over.

This would be particularly helpful to headphone users, as Amiga stereo is headache inducing otherwise.

RowanDDR, 2020-04-26:
"Crossfeed" might be a better term for this feature request than panning.
https://en.wikipedia.org/wiki/Crossfeed

rvalles, 2020-04-26:
Yes, it might be. Panning is what uade123 calls it, for some reason.

marqs85, 2020-04-27:
Digitized audio doesn't go through FPGA, but the audio ADC chip includes a DSP which could be used to perform such mixing.

DavidOliM, 2020-12-27:
Not the same thing but related: play stereo audio as mono.

I have a pretty annoying problem with my ps2 slim that for some reason only the left audio channel plays through the ossc either from component or RGB cables, even if i choose mono in the game's settings. After searching for this problem on google i found someone else with the same problem unsolved. I know it's a stretch since we may be the only two persons experiencing this problem but who knows, it could be useful for someone or something else too.

saturdayinthefart, 2023-02-18:
> 

add a 3rd person in 2023 trying to solve this.

The 3.5m audio cord I'm using is 100% working with stereo sound - tested it with the Line In on my PC, so it's gotta be the OSSC that's bungling it somehow.

Fortunately my main use is on stream, so I can make the audio mono in OBS, but i'd still like proper stereo imaging from the games i'm capturing too :-/

marqs85, 2023-02-19:
@DavidOliM @saturdayinthefart if an audio channel is missing, the board has a defect or has been damaged. Anyway, I just tried configuring the digital mixer of PCM1862 and managed to get it to sum channels into mono (with -6dB gain on each to prevent clipping). There is work ongoing to release v1.xx series firmware, I can add this there as a compatibility option.

marqs85, 2023-07-09:
Mono mode can be now enabled on v1.03 fw. More details on the firmware and its requirements are [here](https://videogameperfection.com/forums/topic/ossc-v1-xx-series-beta-firmware/).

closed 2023-07-09 by marqs85