marqs85/ossc #44: The V./H. Backporch is Actually Front Porch

eatnumber1, 2020-04-07:
When reading through [rtl/scanconverter.v](https://github.com/marqs85/ossc/blob/release/rtl/scanconverter.v) to implement https://github.com/marqs85/ossc/pull/43, I realized that what is coded as `V. Backporch` and `H. Backporch` is actually declaring the duration after the sync period before the active region starts. Specifically, this is viewable starting at [line 905](https://github.com/marqs85/ossc/blob/release/rtl/scanconverter.v#L905) where it says:

```verilog
H_AVIDSTART <= h_config[19:11] + h_config[27:20];   // Horizontal sync+backporch length (0...1023)
...
V_AVIDSTART <= v_config[16:11] + v_config[19:17];   // Vertical sync+backporch length (0...127)
```

... but this is not the back porch, it is the front porch! Was this inversion intentional? Not every console (e.g. [PSP](https://docs.google.com/spreadsheets/d/1LiTGawjTFJEKEKp5twUtanuQmSMtr8vqWUuzMhb--_g)) has equal values for front and back porch, so simply calling the front porch "backporch" is misleading and can produce incorrect output.

marqs85, 2020-04-08:
All the definitions for H/V timings I've seen list the sections in following time-wise order: sync, backporch, active, frontporch.

eatnumber1, 2020-04-08:
Aha, I agree :). After reviewing a few more docs, I was mistaken. Thanks :)

closed 2020-04-08 by eatnumber1