marqs85/ossc #45: 312p 15.63kHz output from Retrotink 2x won't sync on startup

steddyman, 2020-04-09:
When using the Retrotink 2X to process the output of a PAL C64, the OSSC will not sync to the signal at 312p when the C64 is first powered on.

To get it to sync you have to toggle the Retrotink 2X to 624p, then back to 312p.  Then the OSSC syncs perfectly and works just fine.  

You can reset the C64 with a reset button and the signal stays synced.  You can turn the OSSC off and back on and it stays synced.

It is only when the C64 is powered on from cold the OSSC cannot sync.

marqs85, 2020-04-11:
Which device do you use to convert Retrotink 2X output to analog? Can you also list the status information from OSSC (info-button on remote) when it has synced to 312p?

steddyman, 2020-04-11:
Hi, thanks for the response.
I believe I am using this item to convert the HDMI to VGA to feed the OSSC:
https://www.amazon.co.uk/gp/product/B015FIF7XQ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
So do you want the status info when the signal has synced correctly and is working?

marqs85, 2020-04-11:
Yes, the status should be read when it has synced correctly. Have you tried C64->RT2x->HDMI2VGA output into another device so that the HDMI->VGA adapter could be ruled out?

steddyman, 2020-04-11:
I've pressed the info button when synced, and it reports the following:
Prof.1      288p
312p  538674
I am unable to remove the OSSC from the chain because my VGA monitor will not sync at 15khz.
If I set the Retrotink to a line doubled mode, like 625p then the OSSC doesn't show this problem when the C64 is cold booted.
I did contact videogameperfection.com about this about a month ago and the response I got back from Matt that said 'Yeah I noticed that too, we're not sure what causes it and nobody's come
up with a fix unfortunately.'

steddyman, 2020-04-12:
I forgot to mention that when you cold boot the C64 and the OSSC has the issue, the status display on the OSSC alternates between NO SIGNAL and 312P about once per second.

marqs85, 2020-04-13:
The status info looks normal. However, your first post indicates you can power cycle OSSC without issues while other devices are powered on. Toggling RT2x into 624p and back before powering on OSSC for the first time should also work then, right? That would imply there's something different in the HDMI or VGA signal after RT2x has been temporarily set to 624p mode.

steddyman, 2020-04-13:
You are right.  I just tried powering on the OSSC after toggling the RT2x into 624p and back and it did indeed work.  Emm, that would imply the RT2x or the HDMI to VGA converter.
I did connect up the RT2x today direct to the DVI input on my monitor via a mini HDMI to HDMI adapter, and a HDMI to DVI converter and that works though obviously only in 624p mode.

marqs85, 2020-04-14:
In the previous sequence, does it make a difference if you power down the HDMI-VGA adapter (i.e. disconnect and reconnect HDMI) before turning on OSSC? That might give a hint whether the adapter or RT2x outputs inconsistent sync.

steddyman, 2020-04-18:
Sorry for the delay in responding to this.  The Retrotink I always leave powered on.  I've tried disconnecting the HDMI-VGA adapter, then powering on the C64, then re-connecting the HDMI-VGA adapter.  With the OSSC on or Off during this cycle, the result is always the same.  The OSSC style shows the sync problem. I think that also rules out the adapter and only leaves the OSSC has having the startup problem.

I've also just tried powering everything off including the Retrotink, apart from the OSSC.  Then power on the C64 and wait for it to boot, then power on the Retrotink.  Again the OSSC shows the sync issue.

marqs85, 2020-04-18:
It sounds you did not do 624p->312p toggle during the process. Since it seems to be the key for solving the puzzle, I'd start from a situation where everything is powered on and synced correctly. Then power cycle devices one by one and see when the sync cannot be regained without the resolution toggle. Your earlier post indicates that you're already done this for OSSC (which synced OK), so you'd only need to test whether power cycling RT2x or the HDMI->VGA adapter causes sync to be lost.

mikechi2, 2020-04-18:
Sup guys. Thanks marqs for helping with this and sorry for having you spend time on it. I'm pretty sure it is something the new 2X firmware fixes. We're working on that now.

steddyman, 2020-04-18:
Thanks Mike.  PicKit3 ordered.  Please send me over the firmware.  Thanks Marqs for your support so far.

marqs85, 2021-02-27:
Closing as the issue seems related to RT2X and possibly fixed in its newer fws.

closed 2021-02-27 by marqs85