marqs85/ossc #47: 625p is being incorrectly treated as 800×600

ulihox, 2020-04-25:
Feeding AV3 input with 625p signal outputs 800x600.

Info button shows:
Prof.0 800x600
625p 540006

marqs85, 2020-04-26:
Both 800x600 and 576p are ~625-line signals although 800x600 is rarely below 56Hz. The firmware currently doesn't consider refresh rate when selecting sampling parameters and chooses the preset depending on input (800x600 for AV3, 576p for AV1-2), but I could add an explicit refresh rate check for these modes.

ulihox, 2020-04-26:
I've been doing some testing. If I feed into OSSC 576p (50Hz), it outputs 800x600@50Hz, but with 576p (60Hz) it outputs 640x480@60Hz (always through AV3). Is it ok?

Thanks @marqs85.

marqs85, 2020-04-26:
Which device you use to output 576p (60Hz) and how many lines in total it has? There's no such standard so I assume the source reduces the number of blanking lines in addition to increasing pixel clock frequency to achieve 60Hz.

Active area used by the source cannot be decoded from analog video which is why presets based on standard/common modes exist. They are fully customizable, though ("adv. timing"), so you can tune mode parameters if you're not getting what you want. In the original case, you'd need to copy timings from 576p preset to 800x600 preset.

ulihox, 2020-04-26:
I use a dvdo iscan ultra (it only deinterlaces) with a PAL PS2. Most of the games (PAL) work at 50Hz, but some of them allow to use 60Hz.

According to the manual of the dvdo (https://www.manualslib.com/manual/1211743/Dvdo-Iscan-Ultra.html), it outputs 625 lines per frame (576 active).

I'll try to customize the timings.

marqs85, 2020-04-26:
60Hz mode in PAL games is 480i so you should be getting 480p out from dvdo. VESA 640x480 preset is then selected by default since you're using AV3, but you can change it to "DTV 480p" from sampling menu (480p in sampler) to get more optimal horizontal sampling.

marqs85, 2020-04-28:
Fixed in b1892079d8400fdfcd6f26221228c8e04c079ca7.

closed 2020-04-28 by marqs85

Sirethief2525, 2020-05-01:
Hey, I was wondering if one of you could maybe compile the firmware for me
so I could test it out with my psp go? I don't have the knowledge to or
experience to do it myself, so if one of you could help me out that'd be
awesome! Thanks in advance.

marqs85, 2020-06-04:
You can thy [this](http://www.niksula.hut.fi/~mhiienka/tmp/ossc_0.86-pre.bin) test image.