marqs85/ossc #53: Vertical and horizontal movement

filevans, 2020-06-02:
Hello
Please could it be possible to add vertical and horizontal position movement as the screen is out of position on my consoles and the monitor isn't allowing any movement when then OSSC is plugged into it, the options for brightness, contrast and movement are all greyed out (using HDMI to DVI adapter)

marqs85, 2020-06-02:
You should be adjust position via Sampling Opt->Adv Timing->H/V backporch

filevans, 2020-06-02:
Do you know why my monitor won't allow adjustment with the OSSC connected? Brightness/contrast and all geometry is greyed out, I'm using HDMI to DVI as my monitors don't have HDMI. On the other monitor I get "out of range", and also after changing all the back porch and sample length settings, I lost all the settings when I tuned the OSSC off and back on again, so it doesn't save the settings?

marqs85, 2020-06-02:
OSSC runs with a fixed line buffer so its output cannot tear in the traditional sense. Judder is periodically seen on displays which don't frame lock into the incoming frequency. I'm pretty sure the problem is with display (which mark&model?), you could try another one if possible to verify that.

filevans, 2020-06-15:
Yes it only happens on one monitor 
https://iiyama.com/us_en/products/prolite-e2607wsd-1/

It doesn't happen on any other TV or monitor I tried.  Well it's annoying as I wanted to use my 16:10 monitor it has a much better height than 
16:9

closed 2020-06-23 by marqs85