marqs85/ossc #54: Feature request: swap green and blue RCA digitally

omikron24, 2020-06-03:
For whatever reason component RCA order is RGB which is incorrect. The correct format is RBG. This design mistake causes unnecessary cable twist and strain. I could re-solder color connectors however I can't swap the digital processing hence the request. 

Could you please add the option to the "Compatibility" sub menu to swap Green and Blue RCA channels?

marqs85, 2020-06-03:
Unfortunately G/Y line carries sync so the signal path is different and cannot be exchanged with R or B lines. As for why the connector has such order, suitably sized 3xRCA connector part with YPbPr order was not found during design time. The input can also be used for RGsB so it's not exclusively for YPbPr.

omikron24, 2020-06-03:
Will the same issue affect OSSC pro model?

marqs85, 2020-06-03:
Depends whether we can find a suitable 3xRCA connector block with YPbPr color order this time, their availability is not that good these days.

alanbork, 2020-06-03:
the current order never bothered me. You'd have to have a pretty degenerate
cable for it to matter, I think.

omikron24, 2020-06-04:
> 
> the current order never bothered me. You'd have to have a pretty degenerate cable for it to matter, I think.
> […](#)

It's official nintendo cables

omikron24, 2020-06-04:
> 
> 
> Depends whether we can find a suitable 3xRCA connector block with YPbPr color order this time, their availability is not that good these days.

Let me help you with this one as I have the power of google! Jokes aside, I'm not affiliated with any company that makes these but here are some links that may be useful later:

[https://hk-rmt.en.made-in-china.com/productimage/eKSxBAwvyXkI-2f1j00rFzTumGyCHoM/China-AV-Jack-Pin-Jack-RCA-Jack-AV5-8-4-59-.html](url)

[http://www.mfg-components.com/AV%20Pin%20Jack-AC%20Power%20socket-Wire%20to%20Wire%20Connector-Scart%20Socket-Mini%20Din%20Connector/LZ-AV5-01CPK-RCA-CONNECTOR/](url)

Or, single connectors with appropriate colors:
[https://eu.mouser.com/Search/Refine?N=4246260295&Keyword=rca&No=25&FS=True](url)

marqs85, 2020-06-26:
Closing the issue since digital swap not possible on the current HW.

closed 2020-06-26 by marqs85