marqs85/ossc #55: Official firmware .86?

Sirethief2525, 2020-06-05:
Hi, I've been testing a build of .86 that was posted in another issue for the latency tester for awhile now and I know some other changes have been made recently and was wondering when the official .86 firmware might be available. I would just compile the code from the source but I have absolutely no experience with that kind of thing, or if there isn't going to be an official firmware .86 soon could you perhaps build the SD install file for me please? If you could that would be fantastic, thanks for all the work you put into this and I hope to hear from you soon.

eatnumber1, 2020-06-07:
I'm not sure when Marqs will decide to release 0.86, but he did comment with a test build [here](https://github.com/marqs85/ossc/issues/47#issuecomment-639017939). Does that work for you?

Sirethief2525, 2020-06-07:
Yes, sorry I didn't reply sooner. My notifications for stuff like this have
been acting up recently.

marqs85, 2020-06-26:
0.86 released.

closed 2020-06-26 by marqs85