marqs85/ossc #56: Feature Request: 400p PC mode line3x with reduced blanking interval to fit within 162MHz

Darry77, 2020-06-07:
First, I apologize if I am requesting something that is obviously impossible .
The idea would be to decouple the sampling timing from the output timings in order to be able to output 1600x1200 @ 70 Hz in for 400p mode line3x mode without exceeding 162MHz pixel clock .

My OSSC unit cannot handle the current 189MHz pixel clock of line3x mode without artifacting, but, as an example, I have been able to get 1600x1200@70Hz with a 152.75MHz pixel clock from a Geforce FX 5900's Silicon Image TMDS without issue (using an EDID emulator to force that as native mode).

Here are the timings used in my tests : 
[image]

I was hoping something similar might be possible for OSSC .

Thank you in advance .

marqs85, 2020-06-07:
There's on-going work for optimizing line3x timing (see #52), but 180+MHz is tall order for Cyclone IV. Decoupling sampling and output can only be done in a very limited fashion on the current HW, you'll have to wait for the Pro model for better flexibility.

Darry77, 2020-06-07:
Thank you for your response . I will wait and see what can be done, optimization-wise . In the mean time, I will start saving up for an OSSC Pro and hope it comes out soon .

Darry77, 2020-06-08:
I had an idea about an alternative way of doing things that might work on my display and likely others.

What about a custom line2x mode where horizontal sampling is done at 1.6666 times the normal sample rate and no horizontal line multiplication is done?

That would give 1066x800 , which is 4:3 and below 162MHz 

H. Samplerate : 1332 (1332.5)
H. Synclength : 33
H. backporch : 211
H. Active : 1066
V. Synclength : 2
V. Backporch : 34
V. Active : 400

I tried it with line3x mode, which obviously does not give the right aspect, because of the 3x vertical multiplication, but I do get a picture . Although I do get a thin vertical line of a slightly darker shade by the left ede of the image .  This may just be my monitor (Acer VW257) reacting to the 1066x1200 I am feeding it for testing .