marqs85/ossc #57: [Feature Request] OSD indication when changing scanline intensity via remote control

RowanDDR, 2020-06-08:
I noticed that when you increase/decrease scanline intensity via the remote buttons, no OSD indicator appears. I was expecting to see a popup, something like "Scanline intensity xx%". That would be useful, and consistent with other OSD stuff that pops up when you press various remote buttons. Can this be done?

filevans, 2020-06-16:
It does display the intensity when you're actually in that setting in the menu, if that helps

RowanDDR, 2020-06-16:
Thanks filevans, yes I'm aware of that. But still it would be useful to have a popup when arbitrarily pressing the buttons. When you change the volume on your TV you don't expect to have to go deep into the menus to get the volume indicator. Similarly for scanlines, it would be nice to see exactly whats being changed as you change it.

filevans, 2020-06-30:
Yes I agree, it also should have full OSD on the actual TV or monitor screen too in my opinion

marqs85, 2020-10-05:
Implemented in 9c5e7b5.

closed 2020-10-05 by marqs85