marqs85/ossc #62: Super resolution support 2560

filevans, 2020-06-23:
Hello I have my PC outputting super resolutions like 2560x240, and the OSSC is set to x4 which outputs 1280x900. But this squishes the width so you get a shimmering effect / artifacts on the scrolling pixles. So is it not possible for the OSSC / TV to support 2560 width? To avoid this problem. Or is there any work around? Thanks a lot

eatnumber1, 2020-06-23:
Do you know how much overscan is in the horizontal region? 2560x240@60Hz is a 36.8 MHz pixel clock (`2560*240*60`), but that's wrong because the pixel clock math should include the overscan, so it should be `(2560+overscan)*262*60`, where 262 = 240 + vertical overscan. So the pixel clock is _at least_ 40.2 MHz.

The fact that line4x outputs 1280x900 doesn't sound right, it should be trying to do 4x vertical and 4x horizontal, but even if it correctly _tried_ to do it, it wouldn't work. 4x by 4x of 2560x240@60Hz is at least a 643.8 MHz pixel clock (`2560*262*60*4*4`), which is dramatically higher than what the OSSC can support (which is up to around ~162 MHz).

I'm guessing however that the OSSC is attempting to output 1280x900 because that's the [closest mode](https://github.com/marqs85/ossc/blob/4b179d20771fce7dc33c2ae475d601fc1d691487/software/sys_controller/tvp7002/video_modes.h#L120) with a matching vertical line count, although I admit I don't fully understand how the OSSC selects which mode to use among resolutions with identical vertical line counts.

If you want to _only_ line multiply the vertical axis to get 2560x960@60Hz, even that's pushing things as it's a pixel clock of at least 160.9 MHz (`2560*262*60*4`), plus horizontal overscan. To try it out, I think all you need to do is add an optimized mode [such as this one](https://github.com/marqs85/ossc/blob/4b179d20771fce7dc33c2ae475d601fc1d691487/software/sys_controller/tvp7002/video_modes.c#L246) which sets `sample_mult = 1` and `fpga_vmultmode = FPGA_V_MULTMODE_4X`. You probably know this already, but doing this will not preserve the aspect ratio.

Another easier option would be to try increasing H. Samplerate when in 1280x900 mode. I'm not sure how much horizontal overscan is in your 2560x240 image, but overscan + active is what you want to set H. Samplerate to. If you don't know, you can start with H. Samplerate = 2560 and increase from there until the picture is centered and looks right. Displaying a 1px by 1px checkerboard of black+white pixels tends to help with that process quite a lot.

marqs85, 2020-06-23:
The line buffer can only store up to 2048 samples - does your content actually have a width of 2560 or is the pixel clock (horizontal) just multiplied to work around some output limitation?

> The fact that line4x outputs 1280x900 doesn't sound right, it should be trying to do 4x vertical and 4x horizontal, but even if it correctly tried to do it, it wouldn't work. 4x by 4x of [...]

Since the input is analog signal, OSSC has no idea of the horizontal "resolution". With Line4x generic mode, it just samples the signal using "1280x240" preset and multiplies vertical by 4 which then becomes 1280x960 output (perhaps that 900 was a typo, or then the input has less than 262 lines).

filevans, 2020-06-23:
So 2560x960 isn't possible? Thanks

marqs85, 2020-06-23:
> So 2560x960 isn't possible? Thanks

It's not possible without major changes, I'd recommend outputting 1920x240 instead if you want to maximize horizontal resolution.

filevans, 2020-06-30:
Thanks for your help

closed 2020-11-27 by marqs85