marqs85/ossc #64: 5X sync issues on .86

Sirethief2525, 2020-06-26:
So I've been using one of the .86 beta firmwares for sometime now and 5X worked fine, then today I updated to .86 official and for some reason everything seems to sync issues,weird graphical artifacts, and severe clipping or no sound at all when 5X at 1600x1200 since that's the only one that looked right on my TV. I have a TCL-55 4K roku TV if that helps any. Right now 4X seems to be working great and I'm really glad yall added the psp mode to the sampling options.

eatnumber1, 2020-06-26:
Hm, my guess is that it's possible that 45b093d76802554093f6353c738598cf05f3e960 caused timing between the various HDMI_TX pins to differ enough that it's causing problems, even though I didn't see it on my device.

If you're able to, can you post a video of the behavior you're seeing?

Sirethief2525, 2020-06-26:
Yeah, how do I post a video though? I'm not the best with github

Sirethief2525, 2020-06-26:
So here is that video of my snes and n64

Sirethief2525, 2020-06-26:
https://www.youtube.com/watch?v=x3b1WQbxCuM

Here is a link to the video

eatnumber1, 2020-06-27:
Can you try again with this build? This sets fast output on all of the HDMI_TX pins, so may mitigate the problem, but it does introduce timing issues in other places, so it may cause other problems. It should let me know if my theory is right though.

[ossc_0.86-aud-psp.bin.zip](https://github.com/marqs85/ossc/files/4840057/ossc_0.86-aud-psp.bin.zip)

Sirethief2525, 2020-06-27:
I'll test it out today after work and let you know how if fares.

Sirethief2525, 2020-06-27:
Okay so I installed and tested it and now I can't get the optimal timings
to work with 4X or 5X and 5X is stuck on on some weird 16x9 type setting i
can't seem to get to change

Sirethief2525, 2020-06-28:
I found the source of my problem and you're not going to believe it,
apparently my hdmi cable is going out, because as soon as I switched cables
it started working correctly, well at least on the pre-.86 I tested earlier
this month. I'll test final .86 and let y'all know the results.

Sirethief2525, 2020-06-29:
Okay so I tested the official firmware after replacing my cable and can
confirm that it is functioning properly now. Thank-you all for the help in
resolving the issue even though there really wasn't one.

eatnumber1, 2020-06-29:
Glad to hear it. The simple problems are always my favorite because it means it isn’t a ton of work to fix, for which this could have been a tough fix.

Sirethief2525, 2020-06-29:
Yeah I felt super dumb, especially considering small electronics and wiring
repair is my side hobby haha.

closed 2020-07-01 by marqs85