marqs85/ossc #65: 320 / 256 width positions don't match

filevans, 2020-07-15:
Hello

In a CRT if you adjust your horizontal position in the service menu, all the resolutions match. But on the OSSC it doesn't match, i centralised the position for 320x240 and then 256x240 is well out of position. The mega drive uses both of those resolutions for different games for example, and the OSSC uses one setting for all 240p resolutions no matter what width. 

Look how far across one screen is when I have other games exactly right: 

[image]

Something doesn't match properly with the low resolutions and doesn't adjust the position of all low resolutions (240p) equally

marqs85, 2020-08-10:
It might be that the modes have different hsync width which may affect the position on a CRT while OSSC only uses leading sync edge as reference like many digitizers. Need to check sync width on a scope to verify the hypothesis.

marqs85, 2020-08-11:
The sync in 256col mode is slightly shorter according to a measurement. If backporch has identical length in both modes, delay from leading sync edge to picture border is higher in 320col mode which explains shift to the right.

Firebrandx, 2021-07-27:
I recall encountering this and concluding it was a quirk of the console and not the OSSC.

marqs85, 2021-07-27:
The upside with different hsync length is that it's possible to reliably detect which mode MD is outputting (unless the signal goes through a sync conditioner etc.). On OSSC Pro it should be possible to add "MD Auto" sampling preset which would select automatically between existing "MD 256col" and "MD 320col" presets.

Firebrandx, 2021-07-27:
That's actually very nifty!