marqs85/ossc #70: [Feature request] 480p / 576p 16:9 pillarboxed / pillowboxed mode

wolf-re, 2020-10-25:
The idea is to actually produce a 16:9 output image from a 480p (576p) input by [padding it with black pixels left and right](https://en.wikipedia.org/wiki/Pillarbox).

In the 1x (pass-through) mode this should be 854x480 (1024x576) where the 720 / 640 active pixels are sampled to the middle of the buffer. Then the full buffer width is output to yield an actual 16:9 image. This could also work for a 2x mode, with the current line buffer of [2048 pixels](https://github.com/marqs85/ossc/issues/62#issuecomment-648093666) giving 1708x960 (2048x1152).

So if I'm not mistaken this would work nicely for 1:1 pixel aspect ratio cases, like (Dreamcast) 640x480 VGA input. For the non-1:1 cases this would require either undersampling or a slightly wrong AR in the boxed output. However when using the 2x upscale, I believe the OSSC can do the 2x pixels by increasing the ADC sample rate. In this case the non-1:1 AR case would not be undersampled any more and still give a correct AR in the output.

Why would this make sense? Well there seem to be some new-ish TVs (e.g. the Panasonic TX-50EXW784) and receivers (e.g. the Marantz SR6008) which cannot switch aspect ratio in game mode. So everything you blast at them is always displayed in 16:9 regardless of what the source says it should be. This even goes for VGA type input, which makes absolutely no sens as this is clearly 1:1 pixel AR content, but that's what they do. With this feature we could nicely work around such displays.

Would such a signal be anything even remotely spec conform?

Could this be achieved with the current state by creatively using the "Advanced timing tweaker" and "post proc." features?

ps.: I'm volunteering as a tester :smile:  I do have the above mentioned two devices and also a much more compatible desktop monitor.

marqs85, 2020-10-26:
I understand the issue as my Panasonic EZ950 behaves in a similar way (never had the issue with older VT30 plasma, though). However, OSSC has inherent limitation of input sampling rate being coupled to output total width. A few tricks can be done in certain cases (like line3x generic 4:3), but in general you don't have much control over active width if you aim for best compatibility.

For example, let's consider 1440x480 which is probably the likeliest "high-res" 480p mode these devices would accept. Sampling rate would be 2*858=1716 which would have active content on roughly 0.75*1716=1287 pixels. To maintain correct aspect, you'd need to scale that 1287px into pillarboxed area of 1440, i.e. 1080 pixels. It could be possible to do that via nearest neighbor or even linear horizontal scaling on the FPGA, but it'd be a tradeoff in quality.

The upcoming Pro model gets a way with this limitation by decoupling sampling and output. There you can just select suitable sampling rate for target output width which is then scanned out using a different clock.

wolf-re, 2020-10-27:
Ahh ok, thanks for the info!

If I understood correctly the current OSSC could produce such a 1440x480-ish output, just not with the right AR (and sub-optimal sampling)? So if I wanted to test whether my TV accepts that 1440x480 mode, what should the settings be? Something like a Wii in 480p, then the OSSC 480p mode in passthrough, Sample rate 2858, active H pixels 1440 or 1287 ...? I'd also have dreamcast with VGA cable as a 640x480 kind of signal if that fits better.

I'd really like to try that and post my finding here, then at least we all learned something about this "nice" TV :laughing: 

And I'm really looking forward to the pro model (for exactly these reasons), thanks again :+1:

marqs85, 2020-10-28:
On the adv. timing of 480p preset, just double every horizontal parameter and you should get 1440x480.

wolf-re, 2020-10-29:
Yes, so for the record, I tried these settings with a Wii in 480p60 (EDTV) mode:

- 480p in passthrough
- 480 sampler timings
  - H-SampleRate 1716
  - H-Synclen 124
  - H-Backporch 120
  - H-Active 1440
  - rest on defaults

Neither directly connecting this to the TV nor via the receiver gave a picture. I also tried the "Panasonic Hack" and "Full TX setup" and setting the "Default HDMI VIC" to various values (including 2) for both connection paths ... but nope, no picture. I did store these settings in a preset, and reset to defaults, then load preset again, just to make sure TV/RX didn't get confused while I gradually adjusted the settings.

So in any case thanks for your help and I close this feature request as "not feasible on current hardware" :grin:

closed 2020-10-29 by wolf-re