marqs85/ossc #71: 1920x1080 VESA GTF support (H backporch > 255)

puckipedia, 2020-10-25:
The VESA GTF timings for 1920x1080x60hz use a backporch of (2576-2248 =) 328, which is larger than the current (0.86) maximum that can be configured. This means that the display of the Power Mac G4 I have here, which I suspect uses these timings (as i can't find any other example modelines that use 1118 vertical lines), can't be brought into view completely, cutting off the left side a little bit.

```
marisa ~> gtf 1920 1080 60

  # 1920x1080 @ 60.00 Hz (GTF) hsync: 67.08 kHz; pclk: 172.80 MHz
  Modeline "1920x1080_60.00"  172.80  1920 2040 2248 2576  1080 1081 1084 1118  -HSync +Vsync
```

The LCD, connected to the Mac, says: `66.83kHz 59.77Hz`, `1080p 1118-p  451748`

puckipedia, 2021-04-12:
(if others hit this issue: i have a horrible patch at https://github.com/puckipedia/ossc/commit/3d02b503c637f408ce5d9b1faa77559f71e0b539 which adds one (1) extra bit to the backporch, bringing it up to 10 in total. i can't believe that patch worked. run at your own risk, etc, etc.)

marqs85, 2023-06-19:
H.backporch up to 511 is possible in v1.xx series. More details on the firmware and its requirements are [here](https://videogameperfection.com/forums/topic/ossc-v1-xx-series-beta-firmware/).

closed 2023-06-19 by marqs85