marqs85/ossc #76: Color Clamping

Finatra, 2020-11-30:
Hello Marqs,

I had an idea to make the analog to digital conversion even more precise.

If provided with a source's color bit-depth, the result of each sample could be clamped (rounded/snapped/etc) to the closest color within the specified bit-depth.

(If bit-depth is the wrong term, you could also think of it as a sources palette.)

For instance, the Sega Genesis uses a 9-bit color palette, for 512 possible colors.
If a sample measurement falls in between two of those 512 values, then we know it is an error and we can round the result to the closer of the two values.

This would prevent us from outputting RGB values which the input device cannot intentionally generate, and therefore generate a truer conversion.

Let me know what you think!

Thanks for all your work,
Ben

alanbork, 2020-11-30:
Are the colors that off? they look right to me as is. In any case I suspect
there's a lot of issues with this - horizontally there's some blur in the
signal so pixels are not truly independent.  This can actually be a feature
when taken to an extreme - such as transparency effects from blending a
blue checkerboard with other images to make them look underwater.

Finatra, 2020-11-30:
> Are the colors that off? they look right to me as is. In any case I suspect there's a lot of issues with this - horizontally there's some blur in the signal so pixels are not truly independent. This [...]

This feature would only be used in conjunction with optimal timings.

marqs85, 2021-02-27:
Closing as this is essentially duplicate of #42.

closed 2021-02-27 by marqs85