marqs85/ossc #77: Suggestion for OSSC Pro on how to distinguish resolutions with identical timings

Darry77, 2020-12-04:
Sorry if this completely impossible or if this has already been suggested, but I wonder if it would feasible to implement a detection algorithm to automatically switch from 640x400@70Hz to 720x400@70Hz VGA mode sampling by adapting a contrast detection algorithm, as used in digital photography https://en.wikipedia.org/wiki/Autofocus#Contrast_detection ?

IMHO, this would not need to run on every line of every frame, but only long enough to achieve a conclusive match and only often enough to ensure switching when necessary . Some sort of debouncing method to avoid switching continuously would probably be needed to . 

Again sorry, if I am talking nonsense or suggesting the impossible .

marqs85, 2020-12-12:
It's not impossible, but I'd prefer something more robust such as hsync length based detection like 640x480 vs 720x480. Unfortunately 640x400@70Hz and 720x400@70Hz timings are indeed identical (aside from different H. active) as far as I know so there's no reliable indicator which mode is used.

marqs85, 2021-02-27:
Closing since this is a feature request for OSSC Pro.

closed 2021-02-27 by marqs85