marqs85/ossc #78: Feature Request: HDMI CEC/One Touch Play (source switching)

GMCZero1, 2021-01-24:
Greetings
HDMI has a nice feature called One Touch Play, where if a source is active it can send a signal to the receiver/TV to switch to that source. I could be watching TV, then turn a console on and it would instruct my receiver/TV to switch to the appropriate sources automatically.

If this feature were to be considered, there are maybe a couple of ways it could be implemented. One would be to send the signal on power up of the OSSC. The other (the one I'm particularly interested in) would be to detect the presence of sync and send the HDMI control signal if the sync state has changed from no sync to sync detected.

An example of the use case would be if you are watching TV. You switch your SNES on, the OSSC sees the sync signal and in turn sends the HDMI control signal to change to the OSSC source

If this were to be implemented, I would suggest a couple of options:
HDMI control:
1) On sync
2) On power up
3) On power up and sync
4) Off

I am interested in "on sync" but I feel "on power up" may be useful to some people.

Additionally there is "Device OSD Name Transfer" which would allow the OSSC to be seen by the TV/receiver. It is just a small convenience to be able to see the name of the device connected to the input and for the input to be selectable on my TV rather than having to switch to my receiver and select the source there. 

I am no programmer but there may be something useful here
https://elinux.org/CEC_(Consumer_Electronics_Control)_over_HDMI#:~:text=HDMI-CEC%20is%20a%20one,uses%20the%20industry-standard%20AV.&text=It%20was%20defined%20in%20HDMI,to%20control%20CEC-enabled%20devices.

I am happy to test and provide more information on my setup and intended use if this request is of interest

Regards

gbrown128, 2021-01-24:
This is (unfortunately) completley impossible on stock OSSC hardware. There's no connection made to the CEC pin on the HDMI connector.

Would be worth having CEC connected to a GPIO pin on the OSSC Pro I suppose.

GMCZero1, 2021-01-25:
Man, that's a real shame.  It didn't even occur to me that there might be unconnected pins.

I hope this can be implemented on the Pro, seems like a good feature to have.

Now just out of curiosity, would it be possible to connect said pins with a mod wire?  If the feature was implemented in software, is it something an end user could take care of hardware wise?  I'm guessing that now the Pro is a thing, it wouldn't be worth changing the PCB to accommodate CEC

marqs85, 2021-01-26:
I don't think IT6613E even supports CEC, so an extra wire wouldn't help. The HDMI transmitter used in Pro supports CEC and it has been connected on the PCB.

closed 2021-02-27 by marqs85