marqs85/ossc #80: feature request: line4x embedded as 1080p

mistepien, 2021-02-17:
It would be great to have an option on line4x mode which is seen by TV as 1080p. Not every TV works now with line4x mode.

alanbork, 2021-02-17:
tricky since there's not enough memory to buffer anything. But I wonder if
you could use a reduced vertical refresh timing 1080p and just output black
lines during retrace on the input signal?

mistepien, 2021-02-17:
but now line4x has on output 960p -- the point is to add 60 point at top and at the bottom and tag it as 1080p

marqs85, 2021-02-17:
This is not possible with the current model, you need to wait for upcoming Pro HW which is able to implement this feature.

alanbork, 2021-02-17:
it's not as simple as "tagging it" dvi is a real time protocol you have to
put the pixels out at a specific time and rate that matches the pixels
coming in from the console. without a buffer you can't do
arbitrary resizing. But that's why I suggested using reduced vertical
refresh 1080p on the output side. Then you could use the actual vertical
refresh on the input sized to put a black buffer around the top and bottom.

closed 2021-02-27 by marqs85