marqs85/ossc #83: PC-9821 over 24Khz does not look properly

gameblabla, 2021-03-16:
Hello,
just wanted to report that 24Khz on my PC-9821 Nr15 does work but it seems that its resolution is being mis-detected and as result, looks pretty bad.
It's possible to boot in 31Khz mode by pressing Grph + 2 on boot but this disables the internal LCD and some games switch back to 24Khz anyway.

Here's a couple of screenshots taken from my USB capture card with my OSSC connected to it :
[image]
[image]
[image]
[image]

And this is how 31Khz looks like (and how 24khz should look like too)
[image]
[image]
[image]
[image]

Notice the extra black pixels on the right and the bottom getting cropped in Doom 2.

marqs85, 2021-03-16:
Which firmware version are you using? I think some of the PC98 related mode presets got broken in 0.87 (more specifically in b26b213), but they should be now fixed by 2e7141c but it's not yet in official firmware. You might try if v0.86 works until a new fw is released.

gameblabla, 2021-03-16:
> Which firmware version are you using? I think some of the PC98 related mode presets got broken in 0.87 (more specifically in [...]

I'm using 0.88a which seems to predate your commit. I'll see if i can compile it from scratch, if not i'll revert back to 0.86. Thanks

gameblabla, 2021-03-16:
Well Quartus is 7Gb+ so yeah, looks like i won't be able to compile it from source. :/ (If someone could compile me a nightly build, that would be nice)

Flashed 0.86a instead to see what it was like and it's not quite right either.

[image]
[image]
[image]
[image]

Part of the left screen is cropped off and there are still black bars on the right. It does fix the bottom getting cropped up though.

marqs85, 2021-03-16:
You can adjust horizontal position by tweaking H.active in Sampling Opt->Adv. timing.

gameblabla, 2021-03-16:
Have somewhat decent settings with these :
H.samplerate = 848.0
H.synclen = 99
H. backporch = 49
V. synclen = 1
V.active = 400

[image]

It still looks worse than 31Khz though but perhaps that's because my cable is not well shielded ?

marqs85, 2021-07-31:
Closing the the primary issue is fixed in 2e7141c.

closed 2021-07-31 by marqs85