marqs85/ossc #85: Separate sampling options for 240p/288p

filevans, 2021-06-20:
Hello please could you make these use different settings as at the moment they're grouped into the same sampling options and I use different ones for both. I use x2 for 240p and X3 for 288p with different vertical backporch and vertical pixel settings too. 240p is around 60hz and 288p is around 50hz so they should be able to be separated so a separate profile isn't required for each one. Thanks for listening

mistepien, 2022-02-16:
I fully concur. New tv boxes are more and more picky in relation for resolution. I have free-region modded SNES and the only option for NTSC and PAL working for me is 3x with "Generic 4:3". Everything else may work for PAL but not for NTSC (or vice versa). I mean tv says "resolution unknown" or I got wrong proportions. Thus different setting for 240p (NTSC) and 288p (PAL). Would be great. This way I would be able to have 5x for NTSC (240p) and 3x for PAL (288p).

mistepien, 2022-02-16:
Yes, of course but for instance I would be able to use 5x with dejiter in NTSC and 3x with PAL -- that is SNES and it would work very nice. I'm looking forward to have it in my 1.6v OSSC. Once again -- what you are doing is just priceless for retro-community.