marqs85/ossc #86: 256 optimal mode AR option labels update request:

Firebrandx, 2021-07-27:
We need to move away from using semi-misleading terms for square pixels on the 256x240 console modes. "8:7" originally referred to the PAR (pixel aspect ratio) of the Colecovision on up through PS1 for 256 res modes (including the Genesis, NES, and SNES). It has nothing to do with square pixels, and where the confusion comes in is the cropped image of the SNES signal down to 224 lines just so happens to also equal 8:7 AR, but in this case, the pixels are square, and only accounting for 224 lines is not how AR correction (or non-correction) works. Since these consoles can in fact draw graphics in all 240 lines (most can anyway), the actual label for square pixels should be "16:15", not 8:7.

Next issue is the "4:3" label. It's not really 4:3. What it should be called is "+1x H-scale" since that's what the behavior appears to be when I example lossless captures.

Cheers and thanks for considering this!

-FBX

marqs85, 2021-07-28:
The values under "256x240 aspect" option can be a bit confusing indeed. The idea behind 4:3 / 8:7 was to allow selecting intended display aspect ratio when the picture is integer-scaled. Obviously these targets are rarely met 100% due to integer-scaling limitations. Strictly speaking, 8:7 display AR is also correct target only when preset active area is changed to 256x224 as you mentioned.

Perhaps 8:7 should be renamed as "1:1 PAR" similar to what OSSC Pro uses in its scaler mode. That would not tie it into specific resolution, but would be correct for both 240/224 heights or whatever active dimensions user adjusts. While "+1x H-scale" is the behavior for 4:3 in practice, it doesn't really convey the true intent. Maybe "4:3 DAR" would better indicate that the idea is to integer-scale for closest match for 4:3 display aspect ratio. Display aspect ratio itself is still a bit problematic term as output signal may have pillarboxes depending on resolution and thus does not necessarily tell how display itself should present the signal.

Firebrandx, 2021-07-30:
"1:1 PAR" sounds fine too. However, since we're not actually getting "4:3" by adding +1 to the H-scale, I've found it has been misleading several users, and they are surprised when I tell them it's not actually 4:3, nor is it proper DAR correction for these 256 modes anyways (proper formula is 256 * (8/7), which at 4x scale, results in 1170x960).

So if we can't use what the label is actually doing (+1 H-Scale), then a best compromise would be "Pseudo 4:3 DAR".

marqs85, 2021-07-31:
Fixed in 20a5696

closed 2021-07-31 by marqs85