marqs85/ossc #88: Capture card doesn`t show resolutions above 480p correctly

AlivE-git, 2021-11-15:
I have chinese capture card "rullz u3". Yesterday I tested it with gbs-control, and resolutions above 480p work corretly. But my ossc shows this picture in 480p line2x mode. It shows the first quadrant on the picture. Second screenshot shows the same game sceen but in a 480p. I tried with the same capture card pc resolutions such as 960p. It shows correctly.
[image]
[image]
Maybe ossc gives not standart hdmi signal or doesn`t give some metadata?

marqs85, 2021-11-15:
You can try the following:

1. Enable Sampling opt.->Allow upsample2x (disables HDMI pixel repetition)
2. Set Compatibility->Default HDMI VIC to 2 (indicates 480p60, might help with some badly designed sinks)
3. Set Output opt.->TX mode to DVI (disables all HDMI metadata)

AlivE-git, 2021-11-15:
Unfortunately it didn't help.
For the test I tried changing the "480p in sampler" values ​​between "DTV 480p" and "VESA", and the capture card changes the screen resolution (720x480 and 640x480), but it is still "cropped" in x2 mode.

marqs85, 2021-11-15:
If it doesn't work even in DVI mode, then I'm afraid the card just doesn't recognize/support 480pX2 timings. You can try enabling Compatibility->Full TX Setup before changing to the mode and reset the card in OBS. Are the any control options for the card and does it show any detailed information about the signal?

AlivE-git, 2021-11-16:
There are no additional settings, even the resolution is configured in hardware, I can only select the current one in obs.
mjpeg doesn't show any useful information. Maybe there is some kind of program...
[image]
[image]

marqs85, 2021-11-29:
Perhaps the card only supports standard CEA/VESA timings. That would explain why VESA 960p via PC worked. You could confirm this by testing a custom PC modelines such as below:

Horizontal active: 1440
Horizontal front porch: 32
Horizontal sync width: 124
Horizontal total: 1716
Horizontal sync polarity: –

Vertical active: 960
Vertical front porch: 18
Vertical sync width: 12
Vertical total: 1050
Vertical sync polarity: –

Pixel clock: 108MHz