marqs85/ossc #89: Feature Request: add cropping options for line5x / 1920x1080

PunisherHD92, 2021-11-28:
Hello, I'm currently using OSSC (firmware 0.88) on a 1920x1080 monitor.
I've attached my Amiga 500 which outputs 320x240 pixels.
I've understood that the line5x multiplier results in 1200 lines, causing 120 lines to be cropped on my 1080 lines monitor.
From what I see, the cropped lines are both at the top and the bottom of the screen.
But I've noticed that on many Amiga games, there is lot of unused lines at the bottom of the screen.

So, I'd like to be able to choose where the 120 lines are cropped, e.g. having a several choices in "Line5x format", something like:
- 1920x1080 (top cropping only)
- 1920x1080 (top & bottom cropping, same as today)
- 1920x1080 (bottom cropping only)

Alternatively, we could hav a fine cropping option, specifying the exact number of lines we want to crop at the top of the screen, the remaining lines being cropped at the bottom automatically. E.g:
- Top cropping: 10 lines (will result in 110 lines cropped at the bottom)

alanbork, 2021-11-28:
Correct me if I'm wrong, but without a full frame buffer (ie ossc-pro)
that's not really an option - the cropping is happening by your monitor,
not ossc.

marqs85, 2021-11-29:
You can adjust vertical position via  Sampling opt. -> Adv.timing -> V. backporch.

@alanbork number of lines output is fixed (5*262) but only part of those are marked as active (5*224 in this case). Digital monitors only display the active portion.

closed 2021-11-29 by marqs85