marqs85/ossc #91: v0.89 settings export resulting in unreadable SD Card

MrClump, 2022-01-05:
Running a settings export in firmware v0.89 results in an SD card that is not readable by Windows (or Mac). Have tried various sizes and brands of card, including one that was under 1GB in size. Have also tried both erasing all existing partitions using DISKPART CLEAN in Windows, and also pre-formatting the card as FAT16 with the SD Card Association formatter.

I have read one other [report](https://videogameperfection.com/forums/topic/firmware-v0-89/#post-51037) of the the same behaviour.

RowanDDR, 2022-01-06:
Same issue here. Card I'm using is 32GB Kingston micro SDHC card formatted with Tuxera SD Card Formatter 5.0.1.
The same card worked great to update the OSSC firmware from 0.88a to 0.89a, but becomes unreadable/corrupt after a "successful" settings export (and yes, I reformatted the card after the firmware update, before trying settings export).

marqs85, 2022-01-06:
It seems that the partition table / partition generated by export is not recognized by Windows. I'll need to contact the author responsible for the feature to work towards a solution. Formatting the card before export is futile since export itself does it.

paulb-nl, 2022-01-07:
I still have old testing firmware where it still worked so I could compare the exported fat headers. 

On the release firmware the first 3 bytes are zeros but on the earlier version it was 0xEB, 0x00, 0x90. 

The sdcard is readable if I just change the first byte to 0xEB. The third byte doesn't seem to matter.

https://github.com/marqs85/ossc/blob/8068542da1313e1ba874a6e0180ea6fd02ef02f9/software/sys_controller/ossc/fat16_export.c#L29-L31

https://github.com/marqs85/ossc/blob/cbe26114718af0373a5e77b537008402ad26cc01/software/sys_controller/ossc/fat16_export.c#L33-L38

marqs85, 2022-01-08:
Should be now fixed in 101a23b. I also added an option for RAW export which can be used to copy settings directly from ossc to another.

MrClump, 2022-01-08:
Superb. Is it possible to download a compiled version to test?

marqs85, 2022-01-09:
A compiled image is temporarily available [here](http://www.infocult.com/m/tmp/ossc_0.90-aud.bin).

MrClump, 2022-01-09:
> A compiled image is temporarily available [here](http://www.infocult.com/m/tmp/ossc_0.90-aud.bin).

Just downloaded it and flashed my OSSC. The export ran fine, and the resulting .bin file imported into the online profile editor perfectly. Thank you. :-)

mistepien, 2022-03-12:
But 0.89 exported fat16 works fine with linux. There are no partition, filesystem is established directly on "partitionless" device. On the filesystem there is a BIN file.

marqs85, 2022-03-13:
Windows and Mac seem to require those 3 magic bytes (x86 short jump+NOP btw.) to detect the FAT filesystem. Anyway, this has been fixed in v0.90.

closed 2022-03-13 by marqs85