marqs85/ossc #92: Trying to install firmware via JTAG USB Blaster

Cuber445, 2022-01-13:
I am very new to FPGA programming but I've encountered an issue. I'm aiming to install new firmware via JTAG but do not have appropriate files.  I'm looking to convert the .cof file into a .jic, but when i try to open the file in "Convert programming file" I get an error saying "Can't locate programming file ossc.sof (in output_files/&) in Conversion Setup File". How can I go about solving this?

Cuber445, 2022-01-13:
I think I'm just having an issue generating the .sof file. When i open the project I can't seem to compile even the available firmware as i get a warning saying "Script File sys/sysnthesis/sys.qip not found"

marqs85, 2022-01-16:
You have to build qsys output files first - see "Building RTL (bitstream)" section on the README on root. Pre-generated .jic files are also available [here](http://www.infocult.com/m/ossc/fw/jtag_prog/).

closed 2022-02-10 by marqs85