marqs85/ossc #93: Please help (REQUEST)

LilShine, 2022-01-25:
ossc isn't cheap, even eon64 keeps pixels clean with smooth filter, i hate pixels in super mario sunshine wii, and mario 64 rgb mod, can add some filter to erase pixels? ossc is the perfection for me, but just need that filters option, thank you @marqs85 
[image]
[image]
looks like 3d all stars, awesome.

mistepien, 2022-03-15:
but filters mean lag.

pipatron, 2022-05-06:
Sounds like you want the _exact opposite_ of OSSC?

marqs85, 2022-05-07:
Filters don't necessarily increase latency, but within limitations of ossc only a horizontal averaging filter might be practical. Essentially p2_smooth = (p2+p1)/2 where p2 and p1 are current and previous pixel R/G/B values.

pipatron, 2022-05-07:
A similar "horizontal averaging" effect could perhaps be simulated today by enabling the 9 MHz filter and configure the sampler at a much faster rate.