marqs85/ossc #94: Force 4x to be 4:3, not 16:9

mistepien, 2022-02-15:
I have SNES. If use 4x with Generic 4:3 what I got on tv screen is 16:9. It does not happen in case of 3x.

marqs85, 2022-02-16:
3x mode outputs 1280x720 active area with pillarboxes while 4x mode outputs 1280x960. TV aspect ratio should be set to 4:3 for latter but unfortunately not all modern TVs have such option. Generation of a pillarboxed image is more or less a hack that was added for 3x as suitable clocks are available - it's not something that can be used in general.

mistepien, 2022-02-16:
OK, I got it, thanks anyway for OSSC. But the workaround of that limitation is https://github.com/marqs85/ossc/issues/85, Please take a look,

mistepien, 2022-03-11:
What about forcing 1080@50 to be 4:3? Is it also challenging?

mistepien, 2022-03-12:
I managed to find solution. I will leave it here. Koryu Transcoder + OSSC + C64 (PAL) = 4x and output was awful. Very wide. I had to switch on 4:3 aspect ratio in TV -- but then lag was almost 3 times bigger.

Solution: https://videogameperfection.com/forums/topic/adjusting-aspect-ratio/?sfw=pass1647087581
TV is Samsung QE55Q77A. My setting are not perfect: output is not centered and I got white line on the left, but these are not real issues if you have low lag (10ms).

H. samplerare = 1990
H. synclean = 40
H. backporch = 38
H. active = 1920

closed 2022-03-12 by mistepien