marqs85/ossc #96: Not saving H Backporch on 1080p

filevans, 2022-03-28:
Hello, I keep setting H Backporch to 130 and then save it in settings under profile 0 and 1 but when I turn the OSSC back on, the H Backporch has gone back to 148. It only seems to be happening on 1080p and not on other resolutions, not sure if this is a bug with the firmware and if I need to update firmware?
Thank you

marqs85, 2022-03-28:
I recall this was an issue with some older firmware, you should definitely try the latest.

marqs85, 2022-03-29:
Instructions are shown both on the character display and monitor up to the point where the actual update starts. After that status is shown on the character display only.

filevans, 2022-03-29:
Yes but my old firmware didn't have any OSD, anyway the new firmware update has fixed the problem, the settings are being saved! And I really like the OSD too, thanks!!

closed 2023-02-19 by marqs85