marqs85/ossc #97: AUDIO in HDMI with DVI mode

Mutronics, 2022-06-15:
OSSC works through the capture card.
My old 4:3 monitor only works in DVI mode.
When I turn on DVI, I lose sound in the capture card, I turn on HDMI, I get sound, I lose the image.

marqs85, 2022-06-15:
DVI mode does not carry audio. If the capture card is just passing thru the signal without any conversion option, then you could hook analog audio into line-in of your PC while using DVI mode.

Mutronics, 2022-06-29:
Video cards perfectly give sound on DVI.
I think from the software side it is implemented simply.

marqs85, 2022-06-29:
You mean DVI connector or DVI signal? It's possible to send HDMI signal through DVI connector and vice versa, but a standard DVI signal does not carry audio. I don't know if it'd possible to send some kind of partial HDMI mode signal carrying audio but not confusing monitors expecting DVI signal.