marqs85/ossc #98: Amiga 768p resolution through AV3 not supported.

Tallguy297, 2022-07-25:
I have a Commodore Amiga 600 plugged from the RGB port to the OSSC AV1 using a SCART cable.
The Vampire 2.2 accelerator output HDMI plugged into the AV3 of the upscaler.
I have configured the OSSC to auto detect inputs.
This is handy for switching screen modes from ECS to RGB automatically when there is a loss of sync on one of the outputs.

The problem I have is that the AV3 input does not seem to support 1360 x 768 screen resolution.
The maximum mode it seems to handle is 1280 x 720.
If I plug the HDMI output of the Vampire 2.2 directly into my monitor, the higher 1360 x 768 is supported but I cannot autoswitch between screen modes. It's a bit of a pain in the neck.

Is there a solution?

Thank you.

[image]

marqs85, 2022-07-27:
Does the character LCD say NO SYNC when you output those resolutions or are there any signs it's trying to sync? At least 1024 x 768 (60Hz?) should be detected correctly if it's using timings similar to VESA.

Tallguy297, 2022-07-27:
No it doesn't show a NOSYNC. It shows the resolution on the LCD display as 790p.

marqs85, 2022-07-27:
790 lines is a bit off the VESA XGA 60Hz standard (806 lines), but it should be still pick the 1024x768 preset (you can verify this by pressing the INFO key on remote). If the mode works when directly connected to monitor, then it should be just matter of adjusting sync/backporch (via Adv. Timing menu) of regenerated signal to match the source. Is there any information of the exact timings output by Vampire 2.2?

marqs85, 2022-07-27:
Actually, if the output is 790 lines, then I think the issue is with the default backporch 29 which is too much. Try reducing that to e.g. 10 via the Adv. Timing menu.

Tallguy297, 2022-07-27:
Unfortunately the Apollo Team has not been forthcoming with information on timings on the HDMI output of the Vampire 2.2 accelerator card.
I am using standard timings using the CVT utility from Aminet (https://aminet.net/package/driver/video/cvt12) and using the Picasso96TNG tool to provide preferences.

marqs85, 2022-07-27:
The sampling preset on OSSC is selected automatically based on number on input lines etc. While it's not possible to force another preset, you can freely edit timings of the selected preset. In this case it should be enough to reduce V. backporch to 10.

Tallguy297, 2022-08-01:
`Actually, if the output is 790 lines, then I think the issue is with the default V. backporch 29 which is too much. Try reducing that to e.g. 10 via the Adv. Timing menu.`
Thank you, I tried that and worked.

However I have a few more issues...
1024x768@60Hz - Image right hand side disappears off screen.
[image]

1280x720@60Hz - No issues. I had to change the output setting from HDMI to DVI to get this result.
[image]

1360x768@60Hz - Image right & left hand side disappears off screen.
[image]

Any ideas how to fix this problem.

Thank you,
Alexander.

closed 2022-08-01 by Tallguy297