marqs85/ossc #101: Request: Support for 17kHz modes (MDA/Hercules)

root42, 2023-02-11:
It would be cool if IBM MDA and Hercules modes (17kHz) could be supported, too. I can test, as I have the hardware.

I use the GGLabs CGA2VGA converter to bring the TTL levels to VGA levels, which works fine for CGA (15kHz) and EGA (25kHz) modes. However MDA/HGC are still not supported on the OSSC.

marqs85, 2023-02-12:
What do you mean by being not supported, not syncing at all (e.g. no stable numbers on the front LCD) or nothing displayed on the connected monitor? 17kHz @ 50Hz is 340 lines which I think should currently map to GBI preset (while not ideal, should still show something).

root42, 2023-02-12:
It does not sync at all. No numbers on the front LCD. Actually it's 350 lines at 49 Hz, IIRC. At least that's what I measured on the scope on all three cards I tried. 
So there should be something showing you say?

marqs85, 2023-02-12:
I just tested the following generated MDA-like modeline via PC which did work:

`Modeline       "720x340_49 (MDA)" 15.126 720 738 873 882 340 341 344 350 -hsync -vsync`.

The problem might be in sync detection of the analog frontend chip. I'm currently working on a beta firmware which bypasses the chip's sync processing as much as possible and replaces it with custom logic. I'll post soon some details how to take it into use (requires a small HW modification and USB Blaster).

root42, 2023-02-12:
Cool. How do I import this modeline into my OSSC?

marqs85, 2023-02-12:
The modeline was just a sanity check whether a 350-line input signal will match a preset which it did. A new MDA preset could be eventually added on the firmware but current lack of it is not really causing the problem here.

alanbork, 2023-02-12:
49hz sounds pretty weird. @root42 have you tried looking up official docs
on the MDA refresh rate?

root42, 2023-02-12:
@marqs85 FYI, I am connecting through the GGLabs CGA2VGA, which inverts the polarity, so it comes up positive, as expected for VGA, plus it adjusts the signal levels. It just can't do anything about the refresh rate or scanlines, obviously. 
@alanbork Let me check the docs about MDA/HGC!

root42, 2023-02-12:
Docs I can find say 50 Hz. But my cards output 49Hz pretty smack on, according to my scope:

[image]

root42, 2023-02-12:
One more note: The GGLabs SHOULD invert the sync pulses, however I compare VSync between my VGA card (+vsync) to the HGC -- and the HGC always has -vsync! With or without GGLabs! So there is something wrong with my GGLabs! No matter how I set the DIP switches it doesn't invert the vsync. That's probably why my OSSC can't lock on.

marqs85, 2023-02-12:
Wouldn't that mean 18147/49.04=370 lines? Sync polarity shouldn't matter, but I've not really tested much with sources with positive polarity.

root42, 2023-02-12:
370 lines maybe with overscan? I am absolutely no expert regarding video modes... However I get the same numbers on all my cards.

marqs85, 2023-02-12:
Yes, I was talking about total number of lines.

root42, 2023-02-12:
Ok, so you are saying that the sync polarity doesn't matter? Then I guess the GGLabs is perhaps even working as intended. I would have assumed that polarity does matter, but I have no idea how the OSSC works internally. :)

marqs85, 2023-02-12:
The chip is supposed to accept any polarity, but I wouldn't be too surprised if it didn't. [Here](https://videogameperfection.com/forums/topic/ossc-v1-xx-series-beta-firmware/) is the thread for the new beta fw.

wbahnassi-if, 2025-02-15:
I would like to add my voice to this issue. Connecting an MDA card that's adjusted to VGA levels causes the OSSC to go crazy. I tried this on two cards and on two TTL-to-Analog converters, and OSSC was always misbehaving the same way.
It keeps resyncing, the LCD display quickly cycles through various frequencies.. sometimes 18Khz sometimes 32Khz, and of course no picture at all.
I tried adjusting sync thresholds but nothing was able to calm down the OSSC.
I used the same TTL-to-Analog converters to connect EGA to OSSC, and everything worked beautifully (15KHz horizontal frequency).. not sure why 18KHz is so problematic?
My OSSC is on firmware 1.10. I would really like to have MDA working with OSSC. Let me know if there is anything I can do to help narrow down the problem (e.g. probe signals).

marqs85, 2025-02-15:
If you can probe and check the following things, it could provide some insight:
* sync polarities and voltage levels
* vsync leading edge offset from hsync leading edge
* are some hsync pulses missing or irregular, e.g. during vsync?

wbahnassi-if, 2025-02-23:
Took me a little while to order a second probe. Here are the readings:

VSync:
* Positive voltage
* Voltage amplitude 5.2V
* Frequency 49.04 Hz
* Period 20.39 ms

HSync:
* Positive voltage
* Voltage amplitude 5.0V-5.2V
* Frequency 18.4-18.5 KHz
* Period 55.10-55.12 us

VSync Leading Edge offset from HSync:
* Last Hsync during blank to VSync rising edge 9 us
* VSync rising edge to next Hsync rising edge 46.40 us

[image]
[image]

Any missing hsync pulses or irregular?
* Looks like everything is in order during vblank and outside of it. Zoom-in/zoom-out screenshots:
[image]
[image]

marqs85, 2025-02-23:
Vsync length seems to be 16 lines which is out of ordinary. I'd need to test a matching modeline to see if that could be the reason for the detection issues. It also aligned to trailing (not leading) edge of hsync and has negative polarity, but those should not matter.

marqs85, 2025-02-24:
It seems like the video ADC indeed misbehaves with such lengthy VSYNC. It outputs deraged syncs to FPGA despite vsync processing being bypassed (according to datasheet diagram which clearly is false/simplified). It was possible to get functional output with upper level bypass so it seems another compatibility option is needed for OSSC.

marqs85, 2025-02-24:
You can download a test firmware [here](http://www.infocult.com/m/tmp/ossc_1.12-test.bin) and try it with real HW. Just enable "Full VSYNC bypass" from compatibility opt before selecting the input.

wbahnassi-if, 2025-02-24:
Yes, activating the Full VSYNC bypass option fixes the problem and I get a clean and stable display from the MDA card.

marqs85, 2025-03-01:
Option added in fe3c6fdfa77acebfb0a5a73ccdf9f4efa69a04a9 and will be included in next firmware release.

closed 2025-03-01 by marqs85