marqs85/ossc #102: setting file in SD card

jotego, 2023-03-11:
Hello,

I need my OSSC to start up in AV3 mode. I am assuming there is a way to save your settings to a SD card and asking _chat GPT_ told me to use the remote control to save the default settings. Unfortunately, my remote does not work. 

I have been looking for documentation about how to make a settings file in a SD card. But Google and others just send me here or to the PDF manual, which says nothing about it. 

Assuming that OSSC can actually read a settings file from the SD card, could you document this file creation and syntax in the repository, please?

marqs85, 2023-03-11:
Hi,

The settings can be imported from a SD card, but running the import function requires accessing the menu, thus remote control is needed. However, OSSC can be reconfigured to operate via any remote using NEC IR protocol as described [here](https://junkerhq.net//xrgb/index.php?title=OSSC#Remote_control_setup).

The settings are stored/loaded as a binary dump of C struct with some added headers (version etc.). Easiest way to creating settings to SD card is via [this](http://pbnl.byethost7.com/ossc/profiles/) web app (instructions at the end of the page). Importing/exporting settings unfortunately is not very convenient to user as the lack of code space doesn't allow using FAT32, proper parsing etc. These have been addressed on upcoming OSSC Pro project.

jotego, 2023-03-11:
Thank you for taking the time to answer.

I understand then that what I want to do is not possible. The OSSC will always start up to AV1 and I need to use the remote to either change it to AV3 or import a stored setting in the SD card.

I look forward to the upcoming OSSC Pro project.

closed 2023-03-11 by jotego

marqs85, 2023-03-11:
By default OSSC starts from test pattern.  This behaviour can be changed either by modifying the firmware or by adjusting "Settings->Initial Input" on the menu and saving setings, latter obviously needing a remote control.