marqs85/ossc #105: init error -2 - OSSC

duncan1111, 2023-07-04:
i have intermittent error on the screen. 

init error -2

The OSSC will sometimes turn on normally and display the firmware version on the screen. But most times i get init error -2. 

Regardless of of error or normal screen i dont get any display on the screen. No grey background on the tv. 

I can navigate the menu and flash the firmware from the sd card - but i still get the init error -2 or normal screen but never a display on the tv. 

If i connect a scart - OSSC sees the input - but does not display on the tv. 

Iv tried to flash the firmware with the jtag and usb blaster - but it never works - aways goes to fail straight away. 

Any ideas?

marqs85, 2023-07-05:
This error indicates the very first I2C probe fails. The access is made into THS7353 chip, but there may be something else shorting SCL/SDA line into ground. If you don't see test pattern when the board turns on without giving the error, then I'd suspect IT6613 chip is damaged e.g. by incorrect PSU or ESD from HDMI side (should be unlikely with v1.6 or newer boards).

I recommend checking first that there are no weak solder joints on parts connected to SCL/SDA lines. If there's nothing suspicious, then next IT6613 could be removed to see if the board consistently goes past error -2.

duncan1111, 2023-07-06:
No shorts on SCL/SDA - reflowed IT6613 and THS7353 - still error 2.

removed IT6613 and now get init error -4.

will order a replacement IT6613 and THS7353 and post results. 

Thanks

duncan1111, 2023-09-21:
Its now fixed - THS7353 replaced, IT6613 replaced. Then had a short on the 3.3v line and replaced PCM1862. That got it back up and running.

closed 2024-10-06 by marqs85