marqs85/ossc #106: Automatic backporch?

pshirshov, 2023-07-04:
I have several retro machines with different video cards. While OSSC works well for them I have to create individual profiles for each machine, all because of different h.backporch values.

Almost all the VGA displays and scalers (e.g. Gefen scalers) can auto-adjust picture, may it be possible to do the same with OSSC?

marqs85, 2023-07-29:
Automatic position adjustment is a possibility, but as with other scalers its result would much rely on the picture content present during the adjustment (preferably triggered manually when there are bright borders etc. present).

pshirshov, 2023-07-29:
I understand. Though some scalers (e.g. VanCryst) have almost perfect heuristics which never failed (for me). And displays are generally very good at auto adjustment. So it should be possible.