marqs85/ossc #107: Video mode name length

paulb-nl, 2023-10-29:
While working on the profile editor for 1.07 I came across a possible issue.

Some video mode names are 10 characters long which means they have no null terminator. For example "720x400_70": https://github.com/marqs85/ossc/blob/c0dae3da367ab33b85095e55ed75f18d2ee37493/software/sys_controller/ossc/video_modes_list.c#L50

This can cause issues with `vm_display_name` which uses the `strncpy` function.
https://github.com/marqs85/ossc/blob/c0dae3da367ab33b85095e55ed75f18d2ee37493/software/sys_controller/ossc/menu.c#L97

It works without issues now because the `strncpy` function will go past the name and start reading the Vic which for some reason is now stored as a 16 bit value in v1.07 and the high byte is always 0x00.

marqs85, 2023-10-29:
That's a good catch. I increased name size to 11 bytes which did not change the struct since VIC fits into a byte (struct is no more packed as in v0.xx series). The profile structure/version thus does not need to be updated for 1.08 as no new features are planned.

paulb-nl, 2023-10-29:
Unfortunately the struct values are low byte first so the VIC byte starts right after the name.
For example with 1.07 the 480i mode starts with `34 38 30 69 00 00 00 00 00 00` and then `06 00` for the VIC.

So if 1.08 changed the name to 11 bytes then the VIC 06 value will be part of the name and it will read VIC as 00 from a 1.07 profile.

Resetting the profile should fix this. I will also add 1.08 to the profile editor with the 11 bytes name struct.

marqs85, 2023-10-29:
I thought about this possibility before publishing v1.08 and quickly compared 2 exported setting files which were equal before and after the change. Upon further inspection there was no saved profile in place which explains this. People with v1.06 profiles will thus get VIC=0 globally and garbage at the end of 10-char mode name when using 1.08 fw, but nothing more serious.

closed 2024-06-11 by marqs85