marqs85/ossc #108: Support for "LumaCode"?

c0pperdragon, 2023-11-21:
Earlier this year, I came up with the idea of creating a proprietary video format to get digital video data from certain retro computers for further upscaling:  https://github.com/c0pperdragon/LumaCode/wiki/Overview
User feedback is pretty good but I guess for users already owning a OSSC, the mandatory RGBtoHDMI seems a bit overkill. 
So maybe you are interested in adding support for lumacode into the OSSC, as I guess this should be possible with the existing hardware.

marqs85, 2023-11-30:
The idea is definitely interesting and could be supported on OSSC, but to decode multi-sample code sampling rate and phase would need to be adjusted accurately to avoid wrong colors / artifacts.

c0pperdragon, 2023-11-30:
Yes, this will most likely require individual fine tuning, at least for the Atari 8-bit machines what need the highest data rate of all the supported computers. The C64 and and machines with lower data rate normally work out of the box without any re-adjustments. Just by telling the RGBtoHDMI that the source is in fact such a lumacode signal. Of course, I have no idea how to integrate such a feature into your user interface.

Cod3N3rd, 2024-03-08:
Hi,
Did you move forward with this?
I just bought the ossc and took for granted that lumacode was supported.. (obviously wrong).
Looks like lumacode got some traction, at least with atari 8-bit community, and it would be great to see it supported by ossc.

marqs85, 2024-03-13:
This is not yet moved forward, unfortunately. I could create a test version for a configuration that already operates at 2x oversampling, but due to lack of suitable HW and time my debugging capabilities are currently limited.

c0pperdragon, 2024-03-13:
@marqs85 
If this helps, I could extend the firmware on my signal generator to support all lumacode patterns and just send it to you. 
So you would not have to get all those 1980ties machines and also Lumacode-enable them.

marqs85, 2024-03-16:
Let me first create a test firmwared for you to try. If that doesn't work out, then test hardware with pattern generation could be useful.

marqs85, 2024-03-17:
A test firmware can be now downloaded [here](http://www.infocult.com/m/tmp/ossc_1.10-luma.bin). It is configured for C64 so the following settings should be applied:

Input: AV2_RGsB (RCA from C64 connected to green Y jack)
240p/288p proc: Line2x
Line2x mode: 320x240 optim.
H. samplerate: 504 for PAL C64, 512/520 for NTSC depending on VIC-II model
Sampling phase: adjusted manually until colors correct and stable

c0pperdragon, 2024-03-18:
This is totally awesome, that you start to integrate lumacode support into the OSSC. I very much think there is a huge number of existing OSSC users that would love to get this option. Unluckily I do not have an OSSC of my own (went straight to the RGBotHDMI option instead).

But I know of a few prominent users with a github account that have both an OSSC and a VICIIdizer. So maybe someone here can give this a try:
@TheRetroChannel
@JanBeta

I don't know if @randyrossi , who works on the Kawari (a VIC-II replacment for the C64) with possible support for Lumacode output does have an OSSC, but his input would also be very valuable.

TheRetroChannel, 2024-03-18:
I don't have an OSSC, only the tink 5x. But I may be able borrow one to do some testing

marqs85, 2024-03-23:
> I don't have an OSSC, only the tink 5x. But I may be able borrow one to do some testing

That would be great. Let me know if the test firmware works as expected.

TheRetroChannel, 2024-05-23:
@marqs85 Got my hands on an OSSC, but it looks like the above link is dead. Happy to test it out if you repost it

marqs85, 2024-05-23:
The link works for me now, perhaps there was temporary downtime on the server.

TheRetroChannel, 2024-05-24:
Alright, the download worked and I got the OSSC updated. It took me a little while as I didn't realise the new firmwares do not work with the older hardware versions - I think this one was 1.16. Anyway, did the hardware mod and got it all working. But the lumacode output is not working as expected. I can get it to sync and look almost right, but after messing around with the sampling phase and playing with the offset and gain settings I could never get a perfect image.

Here's an example https://youtu.be/UREFcGgG_vU

marqs85, 2024-05-25:
I've updated the test firmware with some improvements, hope that helps.

TheRetroChannel, 2024-05-28:
Thanks marqs85. The new firmware does seem to help with sampling. I can set the phase to 213deg and get a stable image, but the colours are wrong. Playing with the G/Y offset does affect the colours but they never end up correct. I don't know if that's a matter of swapping some palette settings of if that's another issue. Here's another quick video https://youtu.be/EMAyqLI3B1A

Cheers, Mark

TheRetroChannel, 2024-06-01:
Getting close!
[image]

Just the colours are slightly off, especially with red/brown/orange. Here's an example with the same machine, OSSC on the left, s-video on the right:
[image]

Note these were all done on a PAL C64 6569R3 (8 luma level) VIC-II with the following settings:
Video in proc: Lumacode C64
Line2x mode: 384x240 optim.
H. samplerate: 504
Sampling phase: 22deg

I'm not sure if there's a way to adjust the OSSC to output a 4:3 ratio in this mode, or if I just need to set my capture device to squeeze it back down - again I'm still learning my way around the OSSC.

I don't have Lumacode for the Spectrum so can't test that. I do however have both the older 5 luma level VIC-IIs for PAL and NTSC which I can test, the only major one I don't have is the 8 luma level NTSC VIC-II.

marqs85, 2024-06-02:
I used color codes listed [here](https://www.davenicholson.ca/2018/06/commodore-64-color-codes.html) but it seems order of orange and brown there are swapped. I fixed and uploaded the firmware again. You can try e.g. Line5x with 384x240 mode to see if that result to more accurate aspect.

TheRetroChannel, 2024-06-04:
Yep, using 5x with 1920x1200 looks about right. And orange and brown are now correct. 

I think the overall colour palette could still use some tweaking. This one here is the most accurate I've found https://jamontoads.net/p/lumachroma.html (the hex values are towards the end of the page).

marqs85, 2024-06-06:
The palette is now updated with those values.

TheRetroChannel, 2024-06-11:
Finally got a chance to test out the latest update. It looks soo much better - thanks Markus!

For a comparison, svideo on the left, new palette centre, old palette right.

[image]

Of course they don't totally match the s-video output but that's likely more down to my settings with the tink 5x and subtle VIC-II differences. I think the new palette is a pretty damn accurate.

I'll do a few tests with other C64/128 and VIC-II combinations and report back shortly

TheRetroChannel, 2024-06-11:
Alright, this probably won't come as a surprise to those familar with the C64 lumacode. It works with all PAL revisions - long and shortboard, and also works with PAL 128s. I cannot test NTSC as I only have the 6569R56A which does not work correctly with lumacode without a custom firmware from @c0pperdragon at least I assume this is still the case?

As for the rest of the NTSC stuff, I'm unable to test. I guess if we stick with the current palette then it may look a bit weird to NTSC users as the NTSC palette is a bit different. But all the best C64 stuff is coded for PAL so who cares ;-P

marqs85, 2024-06-11:
The core functionality is now integrated in e49106b6358d86cf2e298a3943666a66909e67d1. Closing the issue for now. Support for other systems (like NES) may be added once color encoding details are published.

marqs85, 2024-07-08:
A new test version is [available](http://www.infocult.com/m/tmp/ossc_1.11-lc.bin) with preliminary support for NES added (Line2x or 5x 256col mode). It also restores the more accurate C64 palette. The code was restructured to be able to better support larger amount of systems, hopefully that didn't break Lumacode decoding for existing sources.

TheRetroChannel, 2024-07-09:
C64 colour palette now looks correct. No other issues found there.

NES is working, the colour palette could use some tweaking, but unlike the 64 I don't have any suggestions for this. 

Some pics of SMB1 with composite vs OSSC
Composite
[image]
OSSC
[image]

Also it looks like one of the colour emphasis bits is not working. From what I understand there are 3 emphasis bits that can be set, and according to [this utility](https://forums.nesdev.org/viewtopic.php?t=13264) only the first two are working - see screenshots below.

Colour palette/emphasis tests
Composite - emphasis OFF
[image]
OSSC - emphasis OFF
[image]
Composite - emphasis 1,0,0
[image]
OSSC - emphasis 1,0,0
[image]
Composite - emphasis 1,1,0
[image]
OSSC - emphasis 1,1,0
[image]
Composite - emphasis 1,1,1
[image]
OSSC - emphasis 1,1,1 (looks the same as 1,1,0)
[image]

Everything else seemed to work fine. Again I found the sampling phase works best around 0 degrees. And bumping up the G/Y OFFSET to around 140 removed any noise.

marqs85, 2024-07-09:
NES palette is set to match 2C02 PPU as defined on [this](https://www.nesdev.org/wiki/PPU_palettes#2C02) page. The code did not take into account cumulative effect of multiple emphasis bits which explains the lack of difference between 1,1,0 and 1,1,1. This is now fixed and the linked FW file is updated.

c0pperdragon, 2024-07-11:
I have bin watching this thread with great interest. Nice to see support for lumacode on the OSSC. 

@TheRetroChannel
Mark, can you already confirm that the color emphasis bits are now working? In this case I plan to start selling the PAL variant. 
Making it compatible to the NTSC PPU will take some investigation due to some very strange behaviour in relation to the OAMADDR bug.

TheRetroChannel, 2024-07-11:
Emphasis bits appear to be working with the lastest update.
[image]
I haven't found any other issues apart from the colour palette which I know is very subjective. Personally I like the look of the Kitrinx34 High Saturation, or the YUV palettes from [here](https://github.com/MiSTer-devel/NES_MiSTer/tree/master/Palette).

@c0pperdragon Do you know which palette is being used on the RGBtoHDMI? I think it also looked pretty nice

c0pperdragon, 2024-07-11:
Thank you for confirming that the emphasis bits now work.

I took the palette for the RGBtoHDMI from this wikipedia page: 
https://en.wikipedia.org/wiki/List_of_video_game_console_palettes#Nintendo
But in the end, we should provide multiple palettes to chose from.

wiesl, 2024-07-21:
Can you please also add support for TMS9918A/TMS9928A/TMS9929A versions with TMSdigitizer.

Thnx.

marqs85, 2024-07-21:
It's already supported with MSX/Coleco setting.

wiesl, 2024-07-21:
From a software side, yes.

The hardware version e.g. from https://www.tindie.com/products/c0pperdragon/rgbtohdmi-mono-lumacode/ is missing.

Documentation about Lumacode connection would also be nice.

marqs85, 2024-07-22:
Documentation on the Lumacode HW is found on its own project space, see e.g. [here](https://github.com/c0pperdragon/LumaCode/wiki/TMSdigitizer-(for-ColecoVision-and-MSX)).

OSSC connection and related settings are shortly described in its [wikipage](https://junkerhq.net//xrgb/index.php?title=OSSC#Lumacode).

rattboi, 2024-07-23:
are the C64 lumacode settings appropriate for VIC-20 lumacode as well? I have a basic signal but am having no luck dialing in anything around the 320x240 optimized values given in the wiki for C64 (No mention of VIC-20 there)

wiesl, 2024-07-23:
Thnx. I mean the Lumacode "protocol" itself. e.g. graphics&short description about signals/timing, etc.

marqs85, 2024-07-26:
C64 Lumacode mode should be palette-wise compatible with VIC-20, buth horizontal timings are different (h.total 455 for 6560-101, 497 for 6561-101?).

@wiesl The protocol documentation should be on its own space. Closing this issue since the support has been completed from OSSC side.

Sowden, 2024-10-06:
> This is not yet moved forward, unfortunately. I could create a test version for a configuration that already operates at 2x oversampling, but due to lack of suitable HW and time my debugging capabilities are currently limited.

Not to open this back up.  But is this still the state for Atari 8-bit computers?

marqs85, 2024-10-08:
If someone has the board installed on suitable Atari HW and is able to help with testing & debugging, I can create a test firmware(s) like was done with NES earlier.

Sowden, 2024-10-09:
Cool beans.  I won't be able to get my hands on one till after next year, unfortunately.  I don't have a lot of knowledge for debugging, but I can help testing and giving reports.  I;ll come back here when I have one installed in my machine.  Thanks man.

RamsesUTL, 2024-10-21:
Atari 8-bit would be really great and I would love to help out with some testing, but it seems I have to wait until c0pperdragon starts selling again next year.

Sowden, 2024-10-22:
Yeah, and it seems like selling them next year is right on track.  That gives me a few months to have my OSSC modded so that I can get up to the latest firmware.  I will touch back here and tag @RamsesUTL when I do

Sowden, 2025-01-07:
@marqs85 @RamsesUTL Cool beans.  I got my order in the mail and is on its way to me.  But my local moder that works on my OSSC has been dragging his feet and hasn't made the small firmware upgrade mod yet 😒 I might do the mod myself by the end of the month if this continues.  But you go ahead.  I'll jump in once mine gets in the mail and I'm able to upgrade my firmware.

RamsesUTL, 2025-01-12:
So I successfully tested my NES with the PPUdigitizer and the OSSC. Tomorrow I'll put the GTIAdigitizer in one of my Atari 800XL.

marqs85, 2025-01-12:
A test firmware can be now downloaded [here](http://www.infocult.com/m/tmp/ossc_1.12-test.bin). The following steps need to be done to get GTIA lumacode decoded properly:
* Set ADC PLL BW to High under sync Opt (defaulted to Medium due to some compatibility issues)
* Select some of the following combinations for mode / proc
** Line6x, 320col (recommended)
** Line3x/4x/6x, 384col
** Line2x, 256col
* set H. samplerate to 456
* adjust G/Y offset and phase until stable and correct picture

RamsesUTL, 2025-01-13:
First test looks very promising. I have a stable picture in the Basic screen but still small glitching lines on the self test screen.
More tests tomorrow ...

Sowden, 2025-01-13:
I'm still waiting for my board to come in the mail (and trying to figure out a mod solution).  But I found a program called Atari Control Picture, and it gives a array of color bars that you can use for testing.  Its titled "acp.xex" and you can get it off of the page posted below.  Hope that helps.

http://fujinet.diller.org/ATARI/UTILS/

RamsesUTL, 2025-01-13:
I was about to start looking for something like this, thanks!

marqs85, 2025-01-13:
> First test looks very promising. I have a stable picture in the Basic screen but still small glitching lines on the self test screen. More tests tomorrow ...

So you mean colors were correct but there was just some glitches caused by sampling jitter? I suppose you already tried adjusting PLL BW and phase.

RamsesUTL, 2025-01-13:
Seems like I missed that ADC PLL BW information yesterday and still had sampling jitter then. Today everything looks great.
Atari 800XL with SubCart on LG OLED C1, My settings:
ADC PLL BW -> High
240p/288p proc -> Line6x, Line6x mode -> 320x240 optim., HDMI HDR -> On
H. samplerate -> 456, Sampling phase -> 180 deg
G/Y offset -> 145

[image]

marqs85, 2025-01-14:
Do you know anyone with VCS and Lumacode? It would be good to get that tested too so the issue could be then fully closed.

Sowden, 2025-01-14:
I don't.  But our T. A. C. O. meeting is coming up.  Thats the Texas Atari Computer Owners meeting here in Dallas.  Maybe I could get a 2600 VCS from someone.  you would need to have @c0pperdragon board, the 2600, and an OSSC to fully test it out right?  Maybe C0pperdragon could chime in and help us.

RamsesUTL, 2025-01-15:
Damn, I totally forgot about how I installed a composite video mod in my Atari 2600. That one is soldered to the bottom of the TIA. So I would need to remove that first and then also desolder the TIA. Now thinking about buying a second 2600 to avoid that ....

c0pperdragon, 2025-01-16:
You should be able to see if the TIAdigitizer is working but just plugging its signal into a regular composite video input (if you still have access to a TV that has that). The picture should be stable but monochrome only and with a vertical stripe pattern.

RamsesUTL, 2025-01-16:
Ah, thanks for the hint. I have (unstable, wandering) video over composite of the RT4K. It looks even more distorted on the small CRT I have at hand.

[image]

marqs85, 2025-01-16:
You could also test the composite video into OSSC's Y-jack (unless you already did that when saying you got video). I also just updated the linked firmware image with support for VCS palette added.

RamsesUTL, 2025-01-16:
I'll try the new firmware in a moment but I suspect some kind of problem with my installation or the combination of the TIAdigitizer with my composite mod.

RamsesUTL, 2025-01-16:
New firmware is running but the problem remains that I sadly don't get a stable picture out the TIAdigitizer.
Any idea how to trouble shoot from this point @c0pperdragon ? Should I try to remove the composite mod?

marqs85, 2025-01-16:
If you get it running, the following settings should be dialed for VCS:

* Line4x, 320col
* H. samplerate=456

c0pperdragon, 2025-01-17:
As long as there is no stable picture on a regular TV, there is no hope that the OSSC can do anything with it. 
It is pretty hard to diagnose what is going on just from this picture. Maybe some inputs to the TIAdigitizer are not properly connected. Could be the CSYNC (pin 2) or the clock (pin 4). Another possibility could be some problems with the output driver, but this is quite unlikely. 
In case you have access to an oscilloscope it would be good to check those input pins. Maybe your specific mainboard provides signals that are just a bit too low to be reliably used.

Sowden, 2025-02-07:
Hey ho guys.  So, I've finally gotten my OSSC hardware modded, and finally got my GTIAdigitizer installed in my 800XL.  I soldered the video wire to the RF pin and found ground by just soldering to the RF box itself.  I also cut off the pin to the far right that went into my RF box.  Here is what mine looks like.

[image]

[image]

[image]

I'm like RamsesUTL where I've gotten off to a good start, but I'm not totally there.  I have a picture, but my two problems are that the colors aren't right and I have some vertical rolling noise going upward.  And when I do get color, I have these vertical jailbars.  Here is what mine looks like.

[image]

[image]

Sorry if it's hard to tell the colors from the glare, but the colors are looking pretty bad.  Hey @RamsesUTL, could you upload a json file for your profile for this so that I have a better starting off point?  And any other suggestions would be really great.  Thanks guys.

marqs85, 2025-02-07:
I just updated the test firmware with some small tweaks that should enable VCS support in other LineX modes too (Line5x may not have best results, though).

@Sowden Have you also done the tweaks needed for all Lumacode configurations, i.e. setting ADC PLL BW -> High, adjustment of G/Y offset and sampling phase until picture is stable and has correct colors?

[11 more replies]