marqs85/ossc #109: Request: Support for PC-98 640x400, 24kHz video mode.

QuercusFelis, 2023-12-25:
I've been trying to use the OSSC to translate the signal from a PC9801. It manages to sync the signal under the 640x400@70 input mode, but the picture is offset up and to the left, and I can't seem to correct the timing through any available settings to get the full picture and have it centered. I unfortunately don't understand all the nuances of the OSSC to try and add a mode to correctly interpret the oddball 24kHz video signal myself, but the timing information is thoroughly detailed [here](https://radioc.web.fc2.com/column/pc98bas/pc98dispout2_en.htm). If someone wants to take a shot at this, it would be greatly appreciated, I am totally unable to use the PC98 with any monitor whatsoever, as virtually nothing I can find seems to support the video signal it outputs, so the OSSC is currently my last bet before I start tearing into a CRT to see if I can mess with its timings enough to get a picture.

I'm happy to provide more information as needed.

marqs85, 2023-12-25:
You need to change advanced timing parameters (via sampling opt -> adv timing) to the values listed [here](https://junkerhq.net//xrgb/index.php?title=Optimal_timings) (search for PC-98). Your monitor also needs to support 640x400 or 1280x800 at 56.4Hz.

QuercusFelis, 2023-12-28:
Thanks for the quick reply! Those settings helped a lot. Unfortunately there's still a few lines at the top getting cut off, I'll have to mess around with it more to see if I can fix that, but that seemed to do the trick. I'm wondering if the site I was originally referencing measured the timings incorrectly, because it produced unusable results and differs significantly from the values you referenced.

marqs85, 2023-12-28:
640x400@24 values are not really differing that much if you consider the sums of v_sync+v_backporch (=33) and h_sync+h_backporch (=144).

closed 2024-06-11 by marqs85