marqs85/ossc #111: Loss of synchronization on firmware versions 1.7-1.9

LobAnovich, 2024-03-03:
Since I updated the firmware of my ossc (China 1.6 upgraded) not sequentially, I can’t know exactly which version of the firmware the problems with loss of synchronization started with.  It worked without problems with PS1 on older firmware versions (up to 0.9), after upgrading and installing firmware 1.7, problems began with loss of synchronization (Composite, RGBS, RGB sync on luma) and on version 1.9 they became so frequent that it is simply not possible to use the PS1  .  I tried different video cables and different versions of the console.  All the consoles I tried were region J. Problems in any modes (game menu, BIOS, games (J, U)).

marqs85, 2024-03-03:
Does the issue occur with other consoles than PS1? I have not heard similar reports of PS1 issues with 1.09 firmware, but presumably most are using PAL/NTSC-U consoles. There might be also an issue with the HW upgrade mod or the specific board. You can always downgrade back to 0.9 if that works for you.

LobAnovich, 2024-03-10:
At the moment, I have tested all types of video signals with all possible types of synchronization, different connecting AV and HDMI cables and power supplies, only on one PS 1 (pu18,ntsc j), with which problems arise.  I haven’t changed the settings of ossc itself since installing firmware v.1.x.  Multiplication does not affect the problem.  I can check the same thing on other copies of PS1, as well as Saturn, Dreamcast (All copies of ntsc j are in stock), but for this I will need a little more time.  Are there any recommendations to check any specific settings in ossc?

marqs85, 2024-03-10:
No specific recommendations, just use default settings. Was the 'upgrade' mod verified with any source to ensure the problem is not in the upgrade?

LobAnovich, 2024-03-14:
I have drawn up a check plan, but for now I am far from the place where the device is located.  In a week, I think I will be able to carry out checks according to this plan.  As for the initial malfunction or malfunction during modification, I highly doubt it.  But further tests will show the truth

marqs85, 2024-06-01:
First of all, let's make sure my understanding on results using the firmware versions is correct:
* v1.07-1.09 (after performing the HW mod): issues with 3 different PSX revisions. Saturn, Dreamcast (VGA?) mentioned but not explicitly stated they have the same issue.
* v0.90 (before performing the HW): PSX works OK.

I can only think of 2 causes:
* same sync processing [issue](https://videogameperfection.com/forums/topic/ossc-loosing-av1_rgbs-input-signal-on-firmware-version-1-09) that has been reported with MSX on the latest fw. According to the report only 1.09 is affected, though. This is expected to be resolved eventually once I receive HW that reproduces the issue.
* the China variant is somehow different and incompatible with the HW. Without further details on it, the only feasible fix is to revert to 0.90, possibly also removing the wire added by the HW mod if it's part of the problem.

LobAnovich, 2024-06-01:
> First of all, let's make sure my understanding on results using the firmware versions is correct:
> 
> * v1.07-1.09 (after performing the HW mod): issues with 3 different PSX revisions. Saturn, Dreamcast (VGA?) mentioned but not explicitly stated they have the same issue.
> * v0.90 (before performing the HW mod): PSX works OK.
> 
> I can only think of 2 causes:
> 
> * same sync processing [issue](https://videogameperfection.com/forums/topic/ossc-loosing-av1_rgbs-input-signal-on-firmware-version-1-09) that has been reported with MSX on the latest fw. According to the report only 1.09 is affected, though. This is expected to be resolved eventually once I receive HW that reproduces the issue.
> * the China variant is somehow different and incompatible with the HW. Without further details on it, the only feasible fix is to revert to 0.90, possibly also removing the wire added by the HW mod if it's part of the problem.

I updated the previous post, where I added information that the problem completely disappeared with a return to version 1.07.  But I stayed on version 1.08

LobAnovich, 2024-06-01:
The version with Ali has a simplified power supply circuit and I can, alternatively, assume that the problem may be related to interference in the power circuits.  Although I tested it with several different power supplies, there wasn’t really a very high-quality one among them.  So, were any changes made to firmware 1.08-1.09 that, hypothetically, could depend on the quality of the power supply?

marqs85, 2024-06-02:
No sync related changes were introduced in 1.08, but FPGA fitter seed was changed to meet timing. In the unlikely case that it had some effect on certain boards, I created a test version of the latest [dev firmware](http://www.infocult.com/m/tmp/ossc_1.10-t2.bin) using the seed from 1.07. It doesn't produce identical logic due to other changes, though, but may be still worth a try.

marqs85, 2024-06-05:
Regarding permissible range, do you mean some of the sampling settings do not make sense? 1.10 should ignore saved presets from previous firmwares and use defaults, but if for some reason that does not work you can try explicitly resetting the settings.

LobAnovich, 2024-06-05:
All tests were done on stock settings.  I reset the settings after installing each new firmware.  The range was exceeded when the multiplier was switched.  It’s strange that even in 1x mode the range was exceeded

marqs85, 2024-06-05:
Perhaps I misunderstood, could you elaborate what you mean by "outside the permissible range"? Something reported by the display?

The inconsistency in the results indicate the mod wire is perhaps not properly soldered down, or picking noise somewhere from the PCB.

tecteun, 2024-06-14:
I have _almost_ the same issue, when using anything newer than **1.8**, I get frequent out of sync issues, using in this case a PS2 in PAL format. I've also tried the dev builds that were floating around. The display shows the frequency hopping between 49.9 and 50.0. It happens random, and for about a second, then syncs back to the good frequency (49.9 apparently). I also have seen that it was stable for a longer period. I'm back to 1.8 and that fixes the issue.

So maybe this is related? Or it could be a separate issue.

Anyway, let me know if there is something I can test.

marqs85, 2024-06-20:
This should be now fixed in c91e1ad00c3c12b81dc3ddd234bec3d7fc3be748 which in included in just released v1.10 firmware.

LobAnovich, 2024-06-21:
Thank you for your work and help !
I don't plan to use lumacod, but exporting profiles is important to me.  Is it possible to release a modification of version 1.10 without lumacode, but with working export?  At any time, if I need lumacod, I can flash ossc with standard firmware

LobAnovich, 2024-07-11:
> This should be now fixed in [c91e1ad](https://github.com/marqs85/ossc/commit/c91e1ad00c3c12b81dc3ddd234bec3d7fc3be748) which in included in just released v1.10 firmware.

In general, just as the image disappeared from all my PS1s (ntsc), it continues to disappear.  This happens mainly if the source is interlaced.  Today I connected the pal version of ps2 via a very high quality component cable.  OSSC in the menu shows 625i and the image on the TV periodically disappears.  Previously I connected fpga famicom via vga, the input image is always progressive scan and displaying it on TV has no problems

marqs85, 2024-07-11:
It sounds like your issue is separate from what was reported with MSX, and closer to these below:
https://videogameperfection.com/forums/topic/ossc-and-ps2-intermittent-black-screen
https://videogameperfection.com/forums/topic/wii-eu-launch-model-ossc-1-8-screen-randomly-fades-to-black/

Latest development [firmware](http://www.infocult.com/m/tmp/ossc_1.11-lc.bin) has a couple compatibility options for cases where some displays briefly lose video/audio sync even when OSSC shows no sign of interruption. First you should reduce ADC PLL BW from sync opt to low / ultra low. Additionally you can try setting FPGA PLL BW to low and/or enable Allow TVP HPLL2x from compatibility opt.

marqs85, 2024-08-05:
Closing the issue as the new compatibility options in v1.11 firmware have reportedly fixed similar cases.

closed 2024-08-05 by marqs85